tspopp/AquaMQTT #14: Not issue but feedback

scoudibou, 2024-04-22:
Hello @tspopp ,

Just to tell you I've received my PCB today and installed it on Atlantic Explorer V4 and it's working pretty well.
Thank you!

I've a bit changed the MQTT config you gave me to make a unique device in Home Assistant and had 2 PV control switches  : 

```
mqtt:
  binary_sensor:
    - name: "Heating Element"
      state_topic: "aquamqtt/main/stateElement"
      payload_on: "1"
      payload_off: "0"
      icon: mdi:heating-coil
      unique_id: atlantic_state_heatelement
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
    - name: "Heatpump"
      state_topic: "aquamqtt/main/stateHeatpump"
      payload_on: "1"
      payload_off: "0"
      icon: mdi:heat-pump-outline
      unique_id: atlantic_state_heatpump
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
    - name: "Fan"
      state_topic: "aquamqtt/main/stateFan"
      payload_on: "1"
      payload_off: "0"
      icon: mdi:fan
      unique_id: atlantic_state_fan
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
    - name: "External Boiler"
      state_topic: "aquamqtt/main/stateExtBoiler"
      payload_on: "1"
      payload_off: "0"
      icon: mdi:water-boiler
      unique_id: atlantic_state_extboiler
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
    - name: "Defrost"
      state_topic: "aquamqtt/main/stateDefrost"
      payload_on: "1"
      payload_off: "0"
      icon: mdi:snowflake-melt
      unique_id: atlantic_state_defrost
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
  sensor:
    - name: "Température actuelle"
      state_topic: "aquamqtt/main/waterTemp"
      unit_of_measurement: "°C"
      state_class: measurement
[... 254 lines cut ...]
      state_topic: "aquamqtt/stats/flagPVModeHeatPump"
      payload_on: "1"
      payload_off: "0"
      unique_id: atlantic_ctrl_pvmode_heatpump
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
    - name: "PV Mode Résistance"
      command_topic: "aquamqtt/ctrl/flagPVModeHeatElement"
      state_topic: "aquamqtt/stats/flagPVModeHeatElement"
      payload_on: "1"
      payload_off: "0"
      unique_id: atlantic_ctrl_pvmode_heatelement
      device:
        identifiers: AquaMQTT
        manufacturer: tspopp
        name: AquaMQTT 
        model: V1
```

I have to think how to optimize the usage of the boiler now :)..

Regards

tspopp, 2024-04-22:
Wow, this issue made my day 🥳 🍷. Very happy to see somebody else adopting this project and even testing it successfully against a different branded heatpump 🤩. Kudos! Btw, I think this was very fast, you've received the PCB today, soldered, flashed, installed, tested... Well done.

I need to update the main README, to make some comment regarding the Atlantic Explorer V4 😄 

Regarding the config you have sent, I think I will put it within this repo soon. Of course, you are free to submit a PR as well, if you like. But before merging, I'd like to figure out how homeassistant handles [localization](https://developers.home-assistant.io/docs/internationalization/custom_integration). I think I can provide `de`, `en` ... and I've seen some `fr` in your config? ;)

Cheers

scoudibou, 2024-04-22:
I will add some control, like operation mode.. missing stuffs that you have documented in mqtt.md.
If you need help to translate in french I'll be glad to do it.

scoudibou, 2024-04-24:
#15

closed 2024-04-24 by scoudibou

ALENOC, 2024-11-01:
I also have an Atlantic Explorer but I haven't been so lucky, the messages are discarded, see the image below
[image]
[image] ; do you have modified the protocol?

tspopp, 2024-11-01:
Hey, seems there is a new protocol, there is a ticket for discussion here: https://github.com/tspopp/AquaMQTT/issues/45 

To confirm it is the same new protocol, you need to flash AquaDebug and set the board to Listener Mode and provide a sample of the traces sent to the debug topic. Check out the troubleshooting section for further details

tspopp, 2024-11-09:
Hey @scoudibou, do you might provide the version numbers shown in your HMI engineering menu? I am creating a list which devices in which version numbers have been tested succesfully so far :)

scoudibou, 2024-11-09:
> Hey @scoudibou, do you might provide the version numbers shown in your HMI engineering menu? I am creating a list which devices in which version numbers have been tested succesfully so far :)

It's written Version B.
[image]
Is that you mean ?

tspopp, 2024-11-10:
Yes, many thanks 🙏

ALENOC, 2024-11-24:
Just for information, my Atlantic Explorer is versione E
[image]

tspopp, 2024-11-25:
@ALENOC have you tried [this version](https://github.com/tspopp/AquaMQTT/pull/52), yet? It's not completely finished, but it should parse and provide data to MQTT already.

What kind of Atlantic Explorer is that? v3, v4?

ALENOC, 2024-11-26:
> @ALENOC have you tried [this version](https://github.com/tspopp/AquaMQTT/pull/52), yet? It's not completely finished, but it should parse and provide data to MQTT already.
> 
> What kind of Atlantic Explorer is that? v3, v4?

I confirm that with the new protocol version I can parse something, LISTENER Mode with PSTRH closed[image]

[image]

[image]

[image]

[image]

[image]