tspopp/AquaMQTT #16: Gerber file

Turbox35, 2024-04-25:
Hello, is it possible to provide the gerber file as some PCB suppliers support this format and not kicad?

tspopp, 2024-04-25:
Hey @Turbox35,

i am not familiar with the gerber format. You might want to install [KiCad](https://www.kicad.org/) (it's free!) and export the PCB to the gerber format. I did some quick googling and found a tutorial [here](https://www.pcbway.com/helpcenter/technical_support/Generate_Gerber_file_from_Kicad.html). 

If you successfully created and succeeded with that export, I am happy to receive a PR containing the file to share with others with the same issue :)

Cheers!

Turbox35, 2024-04-26:
Hello @tspopp 
Thanks for the links. I have installed kicad and generated the gerber zip file but as it was new for me I'm not completely sure that was correctly done.
I have ordered PCB with this gerber file, I will let you informed when I will receive it if all is correct and yes, of course, in this case I will share the gerber file.

scoudibou, 2024-04-26:
Here is mine, tested and approuved 👍 
[plot.zip](https://github.com/tspopp/AquaMQTT/files/15129589/plot.zip)

commit 6f247734 (2024-05-01): feat(pcb): add gerber files from @scoudibou, fixes #16
closed 2024-05-01 by tspopp