tspopp/AquaMQTT #18: Thermor aeromax 5 - wiring difference

Turbox35, 2024-05-15:


tspopp, 2024-05-15:
Yes, the current implementation just forwards messages from hmi to main controller and vice-versa. If there is no hmi, the heat pump won't start. 

If only the main connector is connected, you should see information parsed from the main controller within MQTT. This would be a good test without cutting any wires... ;)

Turbox35, 2024-05-15:
You mean connecting the aquamqtt board directly to the HMI unit with external power supply?

tspopp, 2024-05-15:
> You mean connecting the aquamqtt board directly to the HMI unit with external power supply?

> If I pick-up in parallel just the white wire signal and connect it to the board (DHW main side), so like an observer and not an forwarder it will work?

Almost, If you just want to observe the communication from HMI and Main Controller, you should put AquaMQTT in [LISTENER Mode](https://github.com/tspopp/AquaMQTT/tree/main/AquaMQTT) and then connected a parallel wire signal and GND to the main socket of the AquaMQTT Board to make it work:

```mermaid

flowchart LR
    subgraph Heatpump Main Controller Board
        5V_MAIN(5V)
        GND_MAIN(GND)
        DATA_MAIN(DATA)
    end

    subgraph Heatpump HMI Board
        5V_HMI(5V)
        GND_HMI(GND)
        DATA_HMI(DATA)
    end

    subgraph AquaMQTT

        subgraph HmiSocket
            5V_AQ_HMI(5V)
            GND_AQ_HMI(GND)
            DATA_AQ_HMI(DATA)
        end

        subgraph MainSocket
            5V_AQ_MAIN(5V)
            GND_AQ_MAIN(GND)
            DATA_AQ_MAIN(DATA)
        end

    end

    subgraph ExtPower
        5V_EXT(5V)
        GND_EXT(GND)
    end

    ExtPower --- AquaMQTT

    5V_MAIN --- 5V_HMI
    GND_MAIN --- GND_HMI
    DATA_MAIN --- DATA_HMI

    GND_MAIN --- GND_AQ_MAIN
    DATA_MAIN --- DATA_AQ_MAIN
```

Turbox35, 2024-05-15:
Thanks, finally, I have extracted pin from the HMI connector in order to stay with the Man In The Middle mode.

> stats
lwlState = ONLINE
aquamqttMode = MITM
ipAddress = 192.168.3.32
rssiDb = -90
hmi
msgHandled = 8113
msgUnhandled = 0
msgCRCNOK = 0
droppedBytes = 0
msgSent = 16224
main
msgHandled = 18215
msgUnhandled = 0
msgCRCNOK = 0
droppedBytes = 425
msgSent = 9101
activeOverrides = { "operationMode": 0, "operationType": 0, "waterTempTarget": 0, "heatingElementEnabled": 0, "emergencyModeEnabled": 0, "configInstallation": 0 , "time/date": 1 }
overrideMode = DEFAULT
flagPVModeHeatPump = 0
flagPVModeHeatElement = 0
energy
totalHeatpumpHours = 2706
totalHeatingElemHours = 0
totalHours = 2706
totalEnergyWh = 1378209
hmi
waterTempTarget = 62.0
operationMode = MAN ECO OFF
operationType = TIME WINDOW
time = 19:04:42
date = 15.5.2024
emergencyModeEnabled = 0
heatingElementEnabled = 0
antiLegionellaPerMonth = 1
configAirduct = INT/INT
testModeStatus = OFF
configInstallation = HEAT PUMP ONLY
setupState = OK
timerWindowA = 16:15-20:15
timerWindowB = 11:15-15:15
main
waterTemp = 60.4
supplyAirTemp = 21.3
evaporatorAirTempLower = 20.9
evaporatorAirTemp = 21.4
fanSpeed = 0
stateElement = 0
stateHeatpump = 0
stateExtBoiler = 0
stateFan = 0
stateDefrost = 0

It works well and compatible to Thermor Aeromax 5! Thanks again, I will be able to completely remove the cozytouch (the BSB-lan project let me control my Heat pump without it).
Thus, I have cut the HMI wire and make a clean installation.

Is there a possibility to automatically enter the correct date/time using the esp32 when the water hearter is switched ON?

closed 2024-05-15 by Turbox35

tspopp, 2024-05-15:
Nice! This looks great so far! 🍻

> Is there a possibility to automatically enter the correct date/time using the esp32 when the water hearter is switched ON?

Yes, actually the default mode is that the time from the esp32 is used! Notice the active overrides regarding time/date:

`activeOverrides = { "operationMode": 0, "operationType": 0, "waterTempTarget": 0, "heatingElementEnabled": 0, "emergencyModeEnabled": 0, "configInstallation": 0 , "time/date": 1 }`

At this point, the time set within the HMI controller is no longer relevant. Unfortunately it is technically not possible to update/sync the time/date back to the HMI controller.

Let me know if you experience any other issues and we see if we get them sorted! 😄

Turbox35, 2024-05-16:
Hello @tspopp ,
For your information, I created a box for the board because the installation into the Thermor Aeromax 5 is different.
[https://www.printables.com/fr/model/881363-case-for-aquamqtt-with-esp32](url)

tspopp, 2024-05-16:
Nice :+1:

I will put the Thermor Aeromax 5 on main page and link this issue here, for anyone else trying AquaMQTT with that heatpump :+1: