tspopp/AquaMQTT #19: MQTT - publishing frequency

Turbox35, 2024-05-15:
I observe that by default, the ESP32 publish to MQTT under HMI and MAIN sub topic every 0.8s.
Is it raw data frequency rate between HMI and the main board?
Is it possible to choose the publishing frequency to MQTT?

tspopp, 2024-05-16:
Yes, currently the ESP32 publishes a group of messages in case the message from either HMI or MAIN has changed in a single attribute. This leads to lots of MQTT messages and the high frequency you've mentioned.

Actually I wanted to change the behavior in the way that an update is only published in case there is a change to the single attribute. This decreases the frequency and makes it more efficient.

I'd propose to leave this issue open, and we will close it as soon as I made the changes.

Turbox35, 2024-05-16:
Yes of course. In my usage, I would prefer to set a minimum MQTT time sleep. For me, 30s is sufficient between two MQTT messages.

closed 2024-05-28 by tspopp

tspopp, 2024-05-28:
Hey @Turbox35, the latest version contains an entirely different MQTT behavior, which I'll hope you like:

- MQTT messages are only published if the attribute has actually changed, which dramatically decreases the amount of messages published to the broker
- There is a configuration value for a fixed time interval which forces AquaMQTT to update all topics with the current state, by default it is set to 30 minutes:
```
/**
 * Change the time interval where all known attributes are re-published to the MQTT broker.
 */
constexpr uint32_t MQTT_FULL_UPDATE_MS = 1000*60*30;
```

Turbox35, 2024-05-29:
Hello @tspopp 
Oh super! Thanks a lot, I will try it as soon as possible.

I have a personal question about the availability message on mqtt: "online" or "offline".
How the message "offline" is published when the device is not available?

tspopp, 2024-05-29:
Well, this is a functionality of MQTT called "Last Will". If the MQTT broker detects that a client lost the connection, it will publish "the last will" for that client. Read more [here](https://www.hivemq.com/blog/mqtt-essentials-part-9-last-will-and-testament/)

Turbox35, 2024-09-09:
Hello,
I have modified the code in order to reduce temperature measurement noise et decrease the publishing frequency. Moreover it has an effect on the database size.
As an example it publish every 30sec the average temperature of 30 last measurement.

[image]
If you are interested, tell me, I can try to make a PR.

Turbox35, 2024-09-13:
Hello @tspopp perhaps you missed my previous message, are you interested by this implementation?

tspopp, 2024-09-13:
Hey @Turbox35, really appreciate your message and request to help! Actually I see AquaMQTT more as a provider to the raw data provided by the heatpump and currently don't have the intention to do any post-processing with the data on the uC itself. As far as I know, home-assistant will reduce the noise if you add the `state_class: measurement` to your configuration, leading to an average shown in the graphs. Additionaly but a little weird, it seems my Aquawin heatpump does have very limited noise, so it seems they have added some filtering on their heatpumps? Anyway, adding filters would also increase debugging in case issues come up, so I'd always need to check if the additional filtering logic caused the errors. As of now, value sent to mqtt is value actually received on the bus. That is simple to debug.

Of couse you might be able to provide your changes in a fork for users which are intrested in additional filtering :smiley: :+1: 

Cheers!

Turbox35, 2024-09-15:
Hello @tspopp ,

Thanks for your point of view.

On the Domoticz side, there is no effect on the database as Domoticz automatically keeps 1 value by 5min step.
On home assistant, even with state_class: measurement the RW data are kept into the recorder which is set by default to 10days. My unit sends temperature values every second approximately, so I let you imagine why my home assistant database explodes and I don't want to exclude them to the recorder. Perhaps your unit is pre-filtering these temperatures but mine is very noisy, probably due to the low accuracy of the temperature sensors.
Creating a fork is different because forks don't highlight functionality and 99% of aquamqtt user will miss the others possibilities into forks.
Why just let this additional functionality be added as an experimental configuration option which is set to false as default?
For me it's ok as I can modify the code but I'm thinking of the others users who would have the same issue like me.

Looking forward to reading you.

tspopp, 2024-09-17:
Ok, I've been not aware of that database issue. Shot your PR and I we see how we can integrate this feature nicely :+1:

Turbox35, 2024-10-16:
Ohh sorry, I see your release which now filter the temperature sensors data. I didn't take time to make a PR, sorry again @tspopp .