tspopp/AquaMQTT #21: Can not change values

cvhoutvi, 2024-06-01:
Hi TSPOPP,

first of all many thanks for this great project. It is nice to have local control over the Heat Pump Boiler
I build in the PCB in my Atlantic Explorer V4 and I can read all sensors, and states. So at first view everything works.

But when I want to change the target temperature, change the operating mode or activate the PVmode for the heat-pump, nothing happens.
I am in MITM mode and the changes are send to the ctrl-topics, (as you can see in the screenshot of MQTT-explorer below) but the changes are not going to the stats- or hmi-topics, so the changes are not send to the boiler
Even if I publish from within MQTT-explorer to the topics, nothing changes in the stats.

I suppose it is not an error in the code (because it works for others) but a mistake from my side. But I can't find a solution.
First I thought that I have to allow the overrides, but there is no topic to change the override modes.
The "activeOverrides" status confused me. Does a "1" status means that it is allowed to override that value or does it means that the HMI value is overridden with a value send from MQTT? 

Can you point me in the right direction to solve this issue?

thank you very much in advanced.

[image]

scoudibou, 2024-06-01:
Hello,

The active overrides are only to show which parameters are controlled by AquaMQTT, so if you change the target temperature, the waterTempTarget activeOverrides will change to 1. You don't have to set it.

I've no idea why changes aren't applied but just one question : is it normal that your operationMode is ABSENCE ? Does it not disable changes ?

Regards

cvhoutvi, 2024-06-02:
Hi,

thanks Scoudibou for the quick reply. Your answer clarify the purpose for activeOverrides.

The operationMode is another example of a setting that is not accepted. You see on the "ctrl" section that I try to set it to "ABSENCE", but in the HMI section it is still "MAN ECO ON". It does not matter wich operatonMode I choose. It does not take the setting.
The only time that the status changes on stats or HMI mqtt topics is if I make the change on the display of the heat pump boiler. If I change the temp on the HMI display itself (with the rotary knob), it change also on "aquamqtt/hmi/WaterTempTarget" topic. Same happens with changing the operationMode from within the menu on the hmi display itself.
It seems that the board is not listening to the subscriber topics. Do I have to change something in the C++ code to make him listening?

Regards

tspopp, 2024-06-02:
Hey @cvhoutvi,

no, in MITM mode it should listen to the subscribed topics without the need to change any code or config. Maybe it is a bug with the MQTT prefix, according to the screenshot you have "Atlantic/" in your config? You might try something with only lower-cases, but I am just guessing right now.

You might want to share the part of your Configuration.h file, and I will try to recreate the issue.

But since MITM is already working, the fun part is already working :) We will definitely find the issue, why controls are not working yet.

Cheers!

cvhoutvi, 2024-06-02:
Hi @tspopp 

Configuration.h file is in the file attached.

The only change I made in "configuration.h" file is adding a extra GPIO pin (pin 8) and timeout variable (4sec) for the HW watchdog I use. I use that HW watchdog circuit (build around a 555 timer) with most of my electronic projects since it cuts off and restore power from the MCU if no HW trigger is send to the circuit within 5 sec.  I will add that circuit in attachment. Pin 8 of the MCU is the WD-trigger pin. And the +5V-WD is the supply voltage for the MCU board.

So full "configuration.h" is on the first 2 pages of the attached pdf document, the rest of the document consist of all changes I made to your source code. 

Thank you already for the quick response for my question yesterday. It is nice to feel supported :-)

[Aquamqtt own changes to the code(english).pdf](https://github.com/user-attachments/files/15524784/Aquamqtt.own.changes.to.the.code.english.pdf)

[image]

scoudibou, 2024-06-02:
Just some thinkings : 
- Have you tried without watchdog ?
- Have you checked if there is no short circuit with J7 passtrough ?
- Every digital IO of MCU seems to work in 3.3VDC, your WD circuit is in 5VDC, have you already checked that a 3.3VDC of WD-trigger can enable the Q9 to trigger your NE555 ?

tspopp, 2024-06-02:
MQTT subscriptions should be independent of any HW additions you did. I expect the error in the SW. I am able to look into this this evening.

cvhoutvi, 2024-06-02:
Hi,

I found a solution (workaround).
I tested the board for a week with 2 ESPHome devices sending fake 193, 194, 67 messages to the control board and it seems to work, even with the HW watchdog enabled.
But since the control board is connected to the real Heat pump boiler, the HW watchdog is not triggered within the 4 sec. I suppose that the "task" routines takes more than 4 second before giving CPU time back to the "loop" routine.
At this moment I disabled the HW watchdog and all seems to work. I will rewrite the HW watchdog routine as a separate task at a later moment.

My apology for creating this thread. I will close it later. The mistake was all mine 😒 

As I can now change values, I have a last question about it. 
If I change settings form the MQTT topics, is it normal behavior that it does not appear on the device display. For example, if I raise the target temperature to 53°C the display stays on the 50°C if that was the last value set on the display. Same if I enable "ABSENCE" mode, the display stays on "MAN ECO ON"

Again my apology. Since it worked in test environment I did not check the WD when the board was build in.
I would thank you both for the quick response on this topic.  👍

tspopp, 2024-06-02:
No worries. Yes, MITM mode modifies values sent from the HMI controller to the main controller. Since there is no channel / way back to the HMI controller, these changes are not displayed within the original HMI controller. The only true state is provided via MQTT.

closed 2024-06-02 by cvhoutvi