tspopp/AquaMQTT #22: %S (uppercase s) in sprintf()? in MQTTTask correct?

hmgitte, 2024-06-02:
Having problems with the newest code:
What is use of %S (uppercase s) in sprintf()? in file MQTTTask.cpp.
i got messages like this aquamqttSSS <-- S=?
After changing all %**S** to %**s** values in sprintf function, its now possible to receive mqtt messages from aquamqtt. 
Maybe you can check it, my cpp experience realy is not very actual.
My heat pump is an _Austria email explorer evo2_  200 Liter seems now working.
Thanks for your great work.

tspopp, 2024-06-03:
Hey,

in which MQTT topics do you see the issue? %S is correct and used if the string value has been stored using the PROGMEM instruction. Read more regarding [PROGMEM](https://www.arduino.cc/reference/en/language/variables/utilities/progmem/), here regarding [sprintf](https://forum.arduino.cc/t/progmem-and-sprintf/283301/3). 

I don't have any issues with the latest release, so I am wondering what is causing the trouble 😵‍💫 Do you build the project with platformio as described or do you follow an different path?

Since you are the first one with an Austria EMail Explorer Evo2 200 Liter device, did you experience any other northworthy issues or can I just add this machine to the list of proofen machines? :) 

Cheers

hmgitte, 2024-06-12:
Hi,
first of all thanks to your great work.
To compile the design i am using **vscode 1.9**. 
Here the result in mqttexplorer with the original Mqtttask.cpp, with lowercase %s
[image]

Here the same compile after changing all %S to %s, now everything is okay:
[image]

Maybe it is a compiler dependant, but now its okay for me. 
**### yes austria email evo is compatible**. :-)

After a few days the arduino is offline, i have to reboot the whole heat pum, than all is running again.
I don't know why

tspopp, 2024-06-12:
Maybe this is a bug affected by the empty "mqttPrefix" within your configuration. Actually, I never tested with an empty prefix... 🙈 . You might want to check if a defined mqttPrefix in your Configuration will resolve the issue.

In any way, I will check and see If I get this error reproduced. Maybe the weird reset is also linked to this issue. I don't experience any resets in my configuration...

Thanks for giving feedback for the Austria heat-pump, I will add this one to the list :)

tspopp, 2024-06-12:
It seems the "mqttPrefix" I had suspected is not the issue, it runs without any problems if configured with an empty "mqttPrefix". Unfortunately,  I guess we don't even use different compilers, since I also see platformio installed in your VSCode IDE. So I have currently no idea or trace what is causing the reset / broken mqtt messages.

Can you share your configuration (without credentials of course) and maybe other changes you did?

hmgitte, 2024-06-15:
Maybe its helpful... Asking Chatgpt:
Is there a difference between the sprintf function with the parameters %s and %S in the Arduino compiler and the vscode compiler.
Chatgpt:
When comparing the use of the `sprintf` function with the format specifiers `%s` and `%S` between the Arduino compiler (typically based on avr-gcc) and the Visual Studio Code (VSCode) compiler, there are some differences and important points to consider:
### Arduino Compiler (avr-gcc)
- **`%s`**: This specifier is used to output a string (C-string). It expects a pointer to a null-terminated `char` array.
- **`%S`**: In the standard implementation of `sprintf` in `avr-gcc`, there is no `%S` specifier. The `S` is not defined in standard C. Using it would lead to unexpected behavior or an error.

### VSCode Compiler (typically gcc or clang)

- **`%s`**: Here, `%s` is also used to output a null-terminated string.
- **`%S`**: In standard C, as implemented by gcc or clang, there is no `%S` specifier. Using `%S` would typically result in an error because it is not recognized as a valid specifier.

### Summary

- **%s** is a valid specifier in both compilers for outputting strings.
- **%S** is not standardized and is generally not supported in both cases. The exact effect can vary depending on the compiler and environment, but it is likely to either cause an error or lead to undefined behavior.

### Best Practice

To write consistent and portable programs, one should rely on standardized format specifiers. `%s` is the correct specifier for strings in C and is supported correctly in almost all environments. Using `%S` should be avoided as it is not standardized and can lead to issues.

If you are using specific compiler settings or extensions that support `%S`, it is advisable to consult the relevant documentation to understand any deviations and how to handle them.

tspopp, 2024-06-15:
Well, AquaMQTT is actually targeted to a specific platform, which is Arduino (avr-gcc). Therefore utilizing the optimized instructions like `PROGMEM` and `%S` is absolutely fine. 

Do you build and flash using `pio run -e arduino_nano_esp32` ?

Anyway, while revisiting PROGMEM on the official [Arduino documentation](https://www.arduino.cc/reference/en/language/variables/utilities/progmem/), it seems its deprecated for newer Arduino boards. So I will remove the PROGMEM and `%S` portions and we will see if the issues you mentioned will disappear.

tspopp, 2024-06-15:
Please check latest revision on main https://github.com/tspopp/AquaMQTT/commit/03633528489a5889479287be1de52a1e6eb3c191 and provide some feedback if this resolves the issue ;) Thank you!

closed 2024-07-20 by tspopp