tspopp/AquaMQTT #26: Reconnect to wifi

Turbox35, 2024-06-14:
I have an issue about wifi. Indeed, after an firmware update of my router I discovered that the ESP32 didn't reconnect. I have to reboot it to see it again on the network.
I made additionals test, each time I reset my wifi or reboot my router, the ESP32 didn't reconnect.
I have a lot of devices at home (with some esp) and this is the only one with this behaviour.
How is implemented the strategy when the device lost the wifi connexion ?

tspopp, 2024-06-14:
Well, this one is interesting. I thought I fixed automatic WiFi reconnect in https://github.com/tspopp/AquaMQTT/commit/b21f5325526c0ba40fecddf4887106e9fb7e361a

Back then I've found several examples, which suggested doing it this way.  Sine I've never observed any issues, I thought it just works 🤷‍♂️ Will look into this, as soon I have time.

tspopp, 2024-06-18:
I tried to recreate the issue yesterday: I've been playing around with a WiFi hotspot created using my Android phone. The esp is always reconnecting in case WiFi hotspot is disabled and reenabled.

Maybe it is due to the network topology or router, but it is strange, since the auto reconnect behavior seems to be the default arduino esp32 behavior anyway 🤷‍♂️ What kind of router are you using?

Do you use the latest platform framework libraries? Maybe a `pio pkg update` before building helps? If there has been an issue in the arduino framework, a pkg update might resolve it...

Turbox35, 2024-06-19:
Hello @tspopp ,
Thanks for your help.
On my side, I made additionals tests.

If I restart WiFi on my router, it succeeds to reconnect. If I reboot my router, it fails to reconnect. I see nothing on router logs.
For the difference between both WiFi restarting is the time without WiFi, as it takes 20s for WiFi restart and more than 1min for the router reboot.

I have reflashed the esp after an pio pkg update but it changes nothing.

kopierschnitte, 2024-08-17:
Well, unfortunately, I can confirm the behaviour. Today, I needed to reconfigure my Ubiquity stuff and this caused a WiFi downtime of about 30s. Every device reconnected, even plenty of ESP32s (WLED, e.g.) but the AquaMQTT didn't.

Although, my console told me, it was still connected but didn't receive an IP.

According to https://esp32.com/viewtopic.php?f=19&t=39116 it could be a problem that the **WiFi.setAutoReconnect function doesn't work if the station gets "properly" disconnected** (AP sending a disconnect message). It only seems to work if the loss of WiFi appears suddenly and uncontrolled.

One suggests that you could actively monitor the WiFi events and act accordingly. Maybe by introducing a loop or even doing a complete reboot after x sec of no connectivity?

tspopp, 2024-08-17:
Hm, I hoped they address such issues within the official esp32 framework libraries and I don't need an individual hack in a loop to check for WiFi connectivity. Reboot is not an option, we would loose any configuration previously made through MQTT since previous configuration is currently not stored/restored.

I will try re-creating this issue as soon as I have some spare time. I really want to make sure the fix will address the issue and does not cause any other side effects..

tspopp, 2024-08-19:
I was not able to reproduce the issue, but I basically followed the advice do not trust the auto reconnect routine, just as proposed by @kopierschnitte. Therefore I added something customized which is basically sequentially checking for the WiFi connection and attempting a hard reconnect if something is wrong. I've tested this using an Android Hotspot and it was always reconnecting fine. I hope this will finally address this issue. 

I will close this issue as soon as I've merged the PR, but you may feel free to reopen it in case it re-occurs.

closed 2024-08-19 by tspopp

Turbox35, 2024-08-25:
Thanks, I have updated the Arduino with the new code, I will inform you if the problem is solved.

Turbox35, 2024-09-09:
Hello,
After a reboot of the rooter, the arduino is still alive on the network. So, it seems solved.
Thanks a lot and have a nice day.

PR #32 (2024-08-19): fix(wifi): do not trust platform auto reconnect routines
Periodically checking if the WiFi is good instead of trusting the platform auto reconnect routines, since there are edge cases where it seems to fail. Resolves #26