tspopp/AquaMQTT #28: Is it possible to use esp32 wroom or any other esp?

henrykuijpers, 2024-07-06:
I have some esp32 wroom here, can I use it with aquamqtt? 

Or is the esp32 Arduino Nano really needed?

tspopp, 2024-07-06:
Hi @henrykuijpers,

well, I've chosen the Arduino Nano ESP32 since the built quality provided by Arduino is absolutely top notch and everything works just as expected. I had some very fair results with WiFi reliability when playing around with a no name ESP32. Even more, reliability of the Microcontroller is a must, since the heat-pump is out-of-order in case AquaMQTT is not running for any reason.

That said, technically you can alter this project to run on any ESP32 of your choice:

- [ ] Adapt the AquaMQTT PCB to the ESP32 board of your choice using KiCad. Currently it expects a nano layout.
- [ ] Make sure the ESP32 board of your choice allows an input voltage of 5V.
- [ ] Migrate the AquaMQTT project to use `esp-idf` instead the `arduino` framework. Most of the code will likely stay the same, but it definitely needs adaptions for serial, multitasking and wifi parts. This is most likely the expensive task.
- [ ] Adapt the configuration to reflect the pins of your ESP32.
- [ ] Anything else which I might forgot.

As you see, it involves quite a few changes. For me personally, I recommend going with the arduino nano esp32 even if it is a little more expensive ;).

Cheers
Thomas

henrykuijpers, 2024-07-21:
Thanks for this! I'm going with the Arduino Nano ESP32. :) It saves me some hassle.

I did stumble across an issue with the PCB design, not with the design, but with how JLCPCB (and other services) understands the design. https://github.com/tspopp/AquaMQTT/issues/29

Closing this for now.

closed 2024-07-21 by henrykuijpers