tspopp/AquaMQTT #29: Printing PCB - Arduino ESP32 socket part not found at PCB print suppliers

henrykuijpers, 2024-07-20:
Hi,

I'm trying to create a print for the PCB. The "Arduino_Nano_ESP32_Socket" is not found in the jlcpcb parts inventory.

It looks like it should translate to 2x 15p female pin header. But, when exporting the kicad / using the gerber file, this information is not present.

How can adjustments be made, so that this socket is translated to 2x 15p female pin header, so that printing services will understand this and do the right thing?

henrykuijpers, 2024-07-20:
@scoudibou I believe you recently supplied the gerber.zip, which is used to print the PCB. Did you also let your printing service assemble the PCB? Which printing service did you use? And did you also have an issue with this?

tspopp, 2024-07-20:
Not sure, maybe Kicad provides an option to represent the esp sockets as female header. You may also solder the esp32 directly to the pcb, I think that's the reason why there is no female header in the export. If you do the assembly on your own, you are free to solder female headers or the esp32 directly 🤷‍♂️.

Keep us posted and good luck!

scoudibou, 2024-07-20:
hello, I've soldered myself the PCB. For chipset I prefer use header in case of failure or change it's easier.
I used JLCPB and no issue with it. Less than 1 week to receive it (i'm in Europe they're in China).

henrykuijpers, 2024-07-20:
I think indeed you would have to adjust the PCB to have 2x female headers, so that jlcpcb and other services have the notion of what it should look like soldered. 

I'll try to adjust the PCB so that I can submit it that way to jlcpcb. 

Still wondering if I should solder everything myself. Maybe that avoids a lot of potential issues.

henrykuijpers, 2024-07-21:
@scoudibou Why did you solder the PCB yourself? It seems that JLCPCB is super cheap in terms of parts and for soldering as well. They charge almost no money for that, which makes me seem stubborn to solder it myself. Or am I wrong?

I'm trying to replace the Arduino Nano ESP32 footprint in KiCad now with 2x 1x15p pin socket, but it's not easy. This is mainly because I am not familiar at all with KiCad.

tspopp, 2024-10-14:
@henrykuijpers If you have an adapted PCB file which allows the order of pre-soldered pcbs from jlcpcb, I am very happy to receive this as an PR so we can merge and provide it for everyone :+1: Until then, I will close this issue, cause there is nothing left to do or to be answered :)

closed 2024-10-14 by tspopp