tspopp/AquaMQTT #33: do we really need the transistors ?

ouinouin, 2024-09-11:
Hi, i want to give a try for your project, just got a Calypso from atlantic, the HMI looks the same and all the commands listed too.
I want to give a first try by juts tapping on the line and listening, i have level shifters but no transistors, i studied your schematics, and i dont understand the prupose of the darlington  mounted transistors, are they here just to make the signal more strenght ? can i just tap the line and put only a level shifter ? 

What are the labels DHW unknown  on the schematics and the meaning of OWB1/2 ? 
indeed, your project looks great and your documentation effort is impressive . (i m doing th same kind of hack to drive an airton AC unit with tasmota. (berryton on github).

tspopp, 2024-09-11:
Nice, looking forward if we have another heatpump compatible to this project, soon!

If you just want to tap the serial line and listen (LISTENER mode) you don't need the transistors. Level shifting is enough. In case your microcontroller is capable of 5V logic you can even drop the level shifter.

I added the transitors because of this [document](https://ww1.microchip.com/downloads/en/AppNotes/USART-in-One-Wire-Mode-ApplicationNote-DS00002658.pdf) in regards to the _USART in One Wire Mode_ (page 6). This will become important if there is a pair of devices sending and receiving on a single bus. As far as I understood it is basically a protection to not cause any shortage in case both devices are sending (for some unexepected reason) on the bus at the same time.

OWB is "one wire bus" which is actually the data line. DHW unknown is the 4th data line I have observerd within my heatpump which is obviously not needed and just spare. Some heatpumps even just have three wires (data, vin, gnd). I just shorted the "unkown" line in my initial board design, since I was not aware If this data line is in use or not or has some other meaning I am not aware of.

ouinouin, 2024-09-12:
super, thanks for crystal clear answers and for the pointer to the document, it amazes me how complicated devs and electronicians are making just a link to a hmi... instead of pulling rs485 and modbus , with code widely available...
my first plan effectively is to be the listener , so i ll go the wild way and then i ll try to make your board, but also try with other esp32 (a wild thing would be to make your code being an extension of tasmota or esphome :-).

tspopp, 2024-09-13:
I though about esphome and supporting other esp32 devices, too. But the bar of this project is already set high, you need a custom pcb and you need to disassemble the heatpump. If you do these things, you are capable of flashing a device with platformio as well ;) So bringing it to esphome doesn't change much. Using another esp32 device might safe some bucks initially, but in the end you have unexpected issues with WiFi, voltages or whatever. Additionally, you need various PCBs if they don't have a nano layout. Also I guess the user base is still quite low and most users will just use cozytouch for their daily needs. 

So spending more time into lowering the bar will likely have no real benefit and is just not worth the time. To be honest I am quite happy it just runs, doesn't fail and does what it should :). That there are more people happily using AquaMQTT and it is compatibly to other heat pumps is just awesome.

closed 2024-09-13 by tspopp