tspopp/AquaMQTT #34: src/buffer/FrameBuffer.cpp:8:88: error: 'std::string' has not been declared

ouinouin, 2024-09-11:
Hi, got this error while compiling 
apparently according to this  : [https://stackoverflow.com/questions/17040098/c-stdstring-has-not-been-declared-error](https://stackoverflow.com/questions/17040098/c-stdstring-has-not-been-declared-error) , this is not best practice . i know nothing about C/C++ programming so im not judging  that. th workaround mentionned was working  : #include <string> in framebuffer.h

(FYI i m trying to compile for another esp32 board.) (atom M5stack).

tspopp, 2024-09-11:
Yes, true, there is indeed an include missing. As far as I know It depends on the compiler and the order of the files being compiled, if such errors will come up or not during compile time. I will add the include for correctness, even if my build currently is not broken. Thanks for bringing it up :+1:

But since this is a private project, I will just support the platformio build for the Arduino ESP32, which is also tested by the [github actions pipeline](https://github.com/tspopp/AquaMQTT/actions/runs/10460439284/job/28966607067). If you're building for another esp32 board which does not have the arduino framework, it will take some time to port the code. I actually outlined the process of porting this project to another board (without arduino) here: https://github.com/tspopp/AquaMQTT/issues/28#issuecomment-2211764088

closed 2024-09-11 by tspopp

ouinouin, 2024-09-12:
Hi, thanks for answer, im actually compiling with this in the platformio.ini : 
`[env:arduino_nano_esp32]
platform = espressif32
board = m5stack-atom
framework = arduino
test_framework = googletest
build_flags = -std=c++11
# uncomment the below lines to use over the air update
#upload_protocol = espota
#upload_port = 192.168.188.62
lib_deps =
    locoduino/RingBuffer
    FrankBoesing/FastCRC
    PaulStoffregen/Time
    256dpi/MQTT
    adafruit/Adafruit BusIO
    adafruit/RTClib
    Wire
    SPI`
so basically i just changed the board name (and altered the pin names for the RTC.)
i got the same issue further in the code : 
`src/task/MQTTTask.cpp: In static member function 'static void aquamqtt::MQTTTask::messageReceived(String&, String&)':`
`src/task/MQTTTask.cpp: In static member function 'static void aquamqtt::MQTTTask::messageReceived(String&, String&)':`

i find strange that having nearly the same.ini , this kind of error appears , i m unable to compile, i recloned the project this morning and tried to compile without any modif and got another error, ill make a new issue for this one.

commit 397e29c2 (2024-09-11): refactor(framebuffer): remove dead code left from debugging session

- fixes #34