tspopp/AquaMQTT #35: compilation fails vith vanilla config :  too many arguments to function 'esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t*)'

ouinouin, 2024-09-12:
Hi, me again :-) i just split the topic for clarity, 
i tried to compile without any modification  in the code and couldnt success (i guess we got some different versions of arduino or something like that in the platformio setup since the are no versions of the packages defined.
the log states that the platform used is 2024.6.10 

here is the full log , maybe you can paste the log of a successful compilation to see any offending packages version.
```

 *  Exécution de la tâche : platformio run --environment arduino_nano_esp32 

Processing arduino_nano_esp32 (platform: espressif32; board: arduino_nano_esp32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/arduino_nano_esp32.html
PLATFORM: Espressif 32 (2024.6.10) > Arduino Nano ESP32
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, dfu, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.1+sha.371d83e 
 - tool-dfuutil-arduino @ 1.11.0 
 - tool-esptoolpy @ 4.7.3 
 - tool-mklittlefs @ 3.2.0 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-riscv32-esp @ 12.2.0+20230208 
 - toolchain-xtensa-esp32s3 @ 12.2.0+20230208
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RingBuffer @ 1.0.5
|-- FastCRC @ 1.41.0
|-- Time @ 1.6.1
|-- MQTT @ 2.5.2
|-- Adafruit BusIO @ 1.16.1
|-- RTClib @ 2.1.4
|-- Wire @ 2.0.0
|-- SPI @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|-- WiFi @ 2.0.0
Building in release mode
Compiling .pio/build/arduino_nano_esp32/src/main.cpp.o
Compiling .pio/build/arduino_nano_esp32/src/message/MainStatusMessage.cpp.o
Compiling .pio/build/arduino_nano_esp32/src/state/DHWState.cpp.o
Compiling .pio/build/arduino_nano_esp32/src/state/HMIStateProxy.cpp.o
src/main.cpp: In function 'void setup()':
src/main.cpp:48:23: error: invalid conversion from 'uint8_t' {aka 'unsigned char'} to 'const esp_task_wdt_config_t*' [-fpermissive]
   48 |     esp_task_wdt_init(WATCHDOG_TIMEOUT_S, true);
      |                       ^~~~~~~~~~~~~~~~~~
      |                       |
      |                       uint8_t {aka unsigned char}
src/main.cpp:48:22: error: too many arguments to function 'esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t*)'
   48 |     esp_task_wdt_init(WATCHDOG_TIMEOUT_S, true);
      |     ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from src/main.cpp:2:
/home/ouinouin/.platformio/packages/framework-arduinoespressif32/tools/esp32-arduino-libs/esp32s3/include/esp_system/include/esp_task_wdt.h:47:11: note: declared here
   47 | esp_err_t esp_task_wdt_init(const esp_task_wdt_config_t *config);
      |           ^~~~~~~~~~~~~~~~~
*** [.pio/build/arduino_nano_esp32/src/main.cpp.o] Error 1
Compiling .pio/build/arduino_nano_esp32/src/state/MainStateProxy.cpp.o
=================================================== [FAILED] Took 9.20 seconds ===================================================

Environment         Status    Duration
------------------  --------  ------------
arduino_nano_esp32  FAILED    00:00:09.197
============================================= 1 failed, 0 succeeded in 00:00:09.197 =============================================

 *  Arrêt du processus de terminal "platformio 'run', '--environment', 'arduino_nano_esp32'". Code de sortie : 1. 
 *  Le terminal sera réutilisé par les tâches, appuyez sur une touche pour le fermer. 
```

ouinouin, 2024-09-12:
i got into the esp32 doc , apparently the function esp_task_wdt_init() is not taking  an integer as argument but a special type  : 
[doc here from espressif](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/system/wdts.html#_CPPv417esp_task_wdt_initPK21esp_task_wdt_config_t)
i dont know if this function is done differently in arduino.

i commented the watchdog initialisation and could go to the end of  the compilation : 
```
    // initialize watchdog
    //esp_task_wdt_init(WATCHDOG_TIMEOUT_S, true);
    //esp_task_wdt_add(nullptr);
```

tspopp, 2024-09-12:
Hey, well I've looked into this and it is a bit complicated: The error appears on your end, cause you build against a newer [espressif platform](https://github.com/espressif/arduino-esp32/releases) (3.0.1) which breaks the API. My build is still using 2.0.17 . Watchdog needs to be configured differently, you might try with:

```cpp
#include "freertos/FreeRTOS.h"

// (...)

esp_task_wdt_config_t twdt_config = {
    .timeout_ms = WATCHDOG_TIMEOUT_S * 1000,
    .idle_core_mask = (1 << configNUM_CORES) - 1,
    .trigger_panic = true,
};

// (...)

void setup()
{
// (...)
    // initialize watchdog
    esp_task_wdt_init(&twdt_config);
    esp_task_wdt_add(nullptr);
```

I am not sure how you managed to upgrade the framework to a later version, because it seems platformio dropped support for espressif, and with the board `arduino_nano_esp32` I always get a fairly old platform version. Maybe because you were building previously against a different board `m5stack-atom` and it re-uses the framework which has been previously fetched :man_shrugging: 

I managed to upgrade to the latest platform by using https://github.com/pioarduino/platform-espressif32 . I think I will do the upgrade anytime soon, so this build issues should also disappear for you and I am happy to finally get some framework patches, ....

ouinouin, 2024-09-12:
good catch , i dont think platformio is dropping support for espressif, i use a lot the compilation for esphome, and esp-idf seems actively maintained.
in the meantime, you can pin the version of the framework on the platformio.ini 
` platform = espressif32 @ 2.0.17:`

ouinouin, 2024-09-13:
@tspopp 
i found the latest version containing  the 2.0.17 arduino esp32 framework : this is the 6.8.0 : seen [here](https://github.com/platformio/platform-espressif32/releases?page=2).
in fact the 3.2.XX of the arduino : framework-arduinoespressif32 means 2.0.17... : ` - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) `
whereas the framework-arduinoespressif32 @ 3.0.X iss... th framwork 3.0.... what a mess...

so using  `platform = espressif32@6.8.0 ` in he platformio.ini is compiling your code with the old style watchdog 
using `platform = espressif32` is not compiling your code with the old style watchdog
using `platform = espressif32@6.8.1 ` which seems to be a bugfix release is compiling your code with the old style watchdog 
hope this helps

(to what i understand its a good practice to "pin" the version of the framework in the platformio.ini to ensure we re all starting from the same line :-).

ouinouin, 2024-09-13:
indeed , your portion of code compiles with the latest framework.

> ```c++
> #include "freertos/FreeRTOS.h"
> 
> // (...)
> 
> esp_task_wdt_config_t twdt_config = {
>     .timeout_ms = WATCHDOG_TIMEOUT_S * 1000,
>     .idle_core_mask = (1 << configNUM_CORES) - 1,
>     .trigger_panic = true,
> };
> 
> // (...)
> 
> void setup()
> {
> // (...)
>     // initialize watchdog
>     esp_task_wdt_init(&twdt_config);
>     esp_task_wdt_add(nullptr);
> ```

tspopp, 2024-09-14:
Well, I thought moving to v5 will be easy, the first test seemed promising but today I noticed ntp time not working. So I went debugging and also noticed ota is broken, too. So I reverted my initial patch and will prepare the migration to v5 in another branch.

commit cb36cef0 (2024-09-13): feat(framework): upgrade to arduino core 3.0.4 / esp-idf v5.1.4 via pioarduino

- fixes #35
PR #39 (2024-10-14): upgrade(framework): upgrade to latest arduino core / esp-idf
Fixes:
- use https://github.com/pioarduino/platform-espressif32/releases/tag/51.03.05  
- fix broken ntp update
- fix broken ota update due to new behavior of latest framework

Closes #35
closed 2024-10-14 by tspopp