tspopp/AquaMQTT #41: AquaMQTT power consumption is high?

TimSoethout, 2024-10-18:
First, thanks for this board and the software. The installation yesterday was easy and it works great.

I noticed however that from a wall plug power meter that the whole boiler seems to use 4W of energy constantly. Before installing AquaMQTT it was at 0W when idle.

[image]
0W before installing, 4W after.

Is this something that the AquaMQTT can cause? It seems high for an ESP32, although it is sending updates almost every second over wifi. 
It could also be that my measuring plug isn't precise enough, or some threshold is reached... 
I will also try with another plug soon, but maybe this is something explainable?

Note: The reported "AquaMQTT Power Consumed Total" is also always lower than what my plug reports. 
Respectively 0W vs 4W when idle; 
and 585W vs 617W when ending a salmonella run at 62C on heatpump only.
Although this could be explained the internal "AquaMQTT Power Consumed Total" measuring only part of the internal electronics.

tspopp, 2024-10-18:
Nice, it seems this was worth invested time ;)

First regarding "AquaMQTT Power Consumption Total": This value is extracted from the heat pumps protocol and AquaMQTT is not measuring anything, it just passes information it gets from the controller. I am not sure if these consumption sensors in the heat-pump are pre-calibrated by the manufacturer and what is actually measured. Maybe it's just the compressor. In any way, If there is some deviation I can't really do anything about it :man_shrugging:.

In regards to the AquaMQTT power consumption, I doubt that you are able to measure this. There is an esp32-s3 chip on the Arduino Nano ESP32 and if interpret the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-s3_datasheet_en.pdf) correctly, in worst case it consumes about ~1W peak during WiFi communication. Even if we add some losses due to the 5V to 3.3V step-down from the [Arduino Nano ESP32](https://docs.arduino.cc/resources/datasheets/ABX00083-datasheet.pdf) it won't change much. Moreover, the uC is not running at full load, sleeping most of the time ....

kopierschnitte, 2024-10-19:
I guess, the OP meant the actual power consumption "on wire" (wall plug) and not the internal measurements.
My installation doesn't show this behaviour (yet) but in my opinion, it could be theoretically something which causes a wakeup of some components inside the heatpump.

Maybe AquaMQTT is actively sending commands over the bus during this sleep phase. External MQTT commands (maybe a problematic script) could also be the source of the problem?

tspopp, 2024-10-19:
The serial protocol between the controller and the hmi controller is repeated continuously. The messages are always sent in a constant frequency. There are no sleep phases. This is the contract of the serial protocol. AquaMQTT is just holding to the contract, but you may try Listener mode, but I will tell you it won't change anything.

If the observation of this plug was 0 Watts before, that can't be right either. I mean there were always two controllers and even a display running which will consume > 0W. More logical is that a threshold of the plug has been reached. 

If you spot an actually issue, feel free to open a ticket. I don't trace ghost symptoms in my free time.

TimSoethout, 2024-10-19:
OP here. Completely agree. This confirms that it probably isn't related to the ESP. It is more symptomatic of my plug, that probably has a threshold before measuring anything, or isn't precise enough anyway. Thanks for your insights.

I'll close the issue. 
I will continue to document my findings here and will try a different plug.

closed 2024-10-19 by TimSoethout

TimSoethout, 2024-10-30:
A different plug reports 4.1W of power. I couldn't easily test without AquaMQTT. 
But I guess that is reasonable for the whole boiler+AquaMQTT+zigbee switch module (which I still have to remove). 
As discussed above, probably the threshold for the plug was reached at 4W, and power usage is normal and not due to AquaMQTT.

Thanks for all your insights.