tspopp/AquaMQTT #43: Set ventilation

timovd, 2024-10-21:
Would it be possible to send a request to turn on/off the ventilation?
There's a test function, but also a permanent function. Sisi there's low and high. It would be great if this could be done.

timovd, 2024-10-22:
@tspopp Thanks for elaborating. In the installer menu (of my Explorer V4) you can permanently turn on the fan only (low/high). Looking at the names above, I assume these are the settings in the test menu and not the permanent fan. I'll provide pictures later.

timovd, 2024-10-23:
[image]
[image]
[image]
[image]
[image]

tspopp, 2024-10-23:
This is very exciting, cause I don't have this menu entry in my HMI controller. To implement that, I need you to provide a few raw messages of your heat pumps serial protocol. We need to analyze which bytes of serial protocol belong to the fan settings menu you have. In order to do this:

- Modify `Configuration.h` and set `DEBUG_RAW_SERIAL_MESSAGES` to `true`:
```c++

/**
 * Choose to publish raw messages represented as hex-string on debug mqtt topics
 */
constexpr bool DEBUG_RAW_SERIAL_MESSAGES = true;
```
- Reflash AquaMQTT, as usual, make sure to use your own custom configuration for wifi and MQTT broker connectivity.

As soon as you have that, you will have new mqtt topics which are of interest for this issue
- `aquamqtt/main/debug` and `aquamqtt/hmi/debug` which contain the last emitted serial message. You might want to inspect the topic with [mqtt explorer](https://mqtt-explorer.com/).

Now, at your original hmi controller set the mode to "Stop" and copy and paste a pair of the messages, for example:

```
hmi/debug = 23C80040FC00F020F0063C101018FFFF001B573139150000FF0009420000FFFFFFFFFF
main/debug = 250C02EE00EE00F30000000000016441510000002C3E00FFFFFF570942580200100E014E1D
```

Then set it to "Low speed" and copy and paste a pair of the messages.

```
hmi/debug = ...
main/debug = ...
```

Finally set it to "High speed" and copy and paste a pair of the messages.

```
hmi/debug = ...
main/debug = ...
```

--- 
Or, if it is no issue for you running a python script, [ there is a script ](https://github.com/tspopp/AquaMQTT/blob/main/tools/debug.py)which records all the serial traffic. In this case, just record a sequence from Stop to Low to High and attach the traces to the ticket: 

Looking forward :)!

tspopp, 2024-10-23:
I've been looking up the manual after I wrote the instructions, and it seems this settings are only available if you run the heatpump in a different configuration. And yes, after changing the installation mode of my heat pump this menu entry appeared.

So no dumps needed, I will add this as soon as I have free time for this project ;)

timovd, 2024-10-24:
Indeed. I believe the air config should be set to inside/outside.

PR #44 (2024-10-24): feat(fan): allow controlability of airduct and exhaust fan configuration
Allows controlability of Airduct and Exhaust Fan configuration

Fixes #43
closed 2024-10-24 by tspopp