tspopp/AquaMQTT #46: feature request: Donation button

timovd, 2024-10-28:
Is there any way to support the developer with a donation? e.g. Buy me a coffee.

tspopp, 2025-04-10:
Well, I love this ticket 😇. But actually I don't want the extra hassle with the german tax since even donations for open source projects seems to account into revenue and therefore must be declared as such. Therefore I give this project away for free and maintain this for fun as long as it makes me happy.

If you want to still do me favor, just star ⭐ the repo and I will be quite happy 😉

hydex80, 2025-11-22:
> Well, I love this ticket 😇. But actually I don't want the extra hassle with the german tax since even donations for open source projects seems to account into revenue and therefore must be declared [...]

Thanks a lot for your great work. You've got my Star!