tspopp/AquaMQTT #51: Documentation

tukutt, 2024-11-02:
Hi Thomas,
It's not a bug ! But a request to update the documentation. I have a Thermor Aeromax 5.

I created an adapter so as not to cut the original cable.

You can find it here :
https://oshwlab.com/tukutt/aquamqtt-original-pcb-to-rast-co

Are you interesting in a PR to add this connector to the documentation ?

Some pictures :
- https://forum.hacf.fr/t/integration-dun-chauffe-eau-atlantic-explorer-en-mqtt/40493/44
- https://forum.hacf.fr/t/integration-dun-chauffe-eau-atlantic-explorer-en-mqtt/40493/58

Thanks

tspopp, 2024-11-03:
Nice 🎉, sure let's add this! Do you have any other issues with the Thermor Airomax 5? Are the debugging stats values ok, so almost no dropped bytes in both HMI and Main channel?

Maybe you can also provide the software version shown in your HMI and Main Controller, since there is a new protocol version which is not yet supported. And I would love to determine at which version the current Aquamqtt implementation gets incompatible ;)
https://github.com/tspopp/AquaMQTT/issues/45#issuecomment-2440025043

tukutt, 2024-11-03:
Thanks !

User Interface : E
Control : G

[image]

[image]

Just 1 byte drop when I started it.

tukutt

tukutt, 2024-11-04:
Of course! Give me a few days to make the log.

closed 2024-11-11 by tspopp

tukutt, 2024-11-11:
Sorry for the delay. This some debug log. I hope it helps you
[aquamqtt_hmi_debug_hex.csv](https://github.com/user-attachments/files/17707661/aquamqtt_hmi_debug_hex.csv)
[aquamqtt_hmi_debug_dec.csv](https://github.com/user-attachments/files/17707662/aquamqtt_hmi_debug_dec.csv)
[aquamqtt_main_debug_hex.csv](https://github.com/user-attachments/files/17707663/aquamqtt_main_debug_hex.csv)
[aquamqtt_main_debug_dec.csv](https://github.com/user-attachments/files/17707664/aquamqtt_main_debug_dec.csv)

PR #54 (2024-11-11): doc(devices): rast adapter, list of devices with sw version
Fixes #51 

@tukutt The reference to your adapter is a proposal. If you like it, we merge it :) else give me feedback and we change accordingly ;)