tspopp/AquaMQTT #56: Support for ESP32-S3

xtrmenl, 2024-11-25:
i just got this working on a Cheap ESP32-S3 nano (8MB version), which is pin compatible, but less then half the price of the arduino 

`[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
framework = arduino
build_flags = -std=c++11
lib_deps = 
	locoduino/RingBuffer@1.0.5
	FrankBoesing/FastCRC@1.41.0
	PaulStoffregen/Time@1.6.1
	256dpi/MQTT@2.5.2
	adafruit/Adafruit BusIO@1.16.1
	adafruit/RTClib@2.1.4
	Wire@3.0.5
	SPI@3.0.5
	https://github.com/denyssene/SimpleKalmanFilter.git#v0.2
	ArduinoJSON@7.2.0`

to be added to the Platformio

tspopp, 2024-11-25:
You might want to share a link to the esp32 you bought and I will add this to the instructions. I can also add another environment to the platformio.ini. Thanks for leaving this comment :+1:

xtrmenl, 2024-11-26:
the exact part is this one 

https://www.aliexpress.com/item/1005007291319086.html
ESP32-S3-Nano Development Board Based on ESP32-S3R8 For Arduino

on the pcb it states waveshare esp32-s3-nano
as aliexpress links do not always last this is the manufacturer site
https://www.waveshare.com/wiki/ESP32-S3-Nano

closed 2025-01-19 by tspopp

BmdOnline, 2025-01-19:
I managed to use this board, but with some changes.
Add these lines to  `platformio.ini`.
Pay attention to `build_flags` directives  `-DARDUINO_USB_CDC_ON_BOOT=1` and  `-DARDUINO_USB_DFU_ON_BOOT=1`

```ini
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
framework = arduino
build_flags =
  -std=c++11
  -DARDUINO_USB_CDC_ON_BOOT=1
  -DARDUINO_USB_DFU_ON_BOOT=1
#upload_protocol = espota
#upload_port = 192.168.188.62
lib_deps =
    locoduino/RingBuffer@1.0.5
    FrankBoesing/FastCRC@1.41.0
    PaulStoffregen/Time@1.6.1
    256dpi/MQTT@2.5.2
    adafruit/Adafruit BusIO@1.16.1
    adafruit/RTClib@2.1.4
    Wire@3.0.5
    SPI@3.0.5
    https://github.com/denyssene/SimpleKalmanFilter.git#v0.2
    ArduinoJSON@7.2.0
```

Then, I have to change pins definition into `Configuration.h`.
Instead of D2 to D5, I have to use GPIO #
```c++
constexpr uint8_t GPIO_MAIN_RX = 5;
constexpr uint8_t GPIO_MAIN_TX = 6;
constexpr uint8_t GPIO_HMI_RX  = 7;
constexpr uint8_t GPIO_HMI_TX  = 8;
```

prouserrail-hub, 2025-11-07:
For those wondering: 
I bought the unit from Ali (ESP32-S3 Nano) ca Oktober 2025.

Got it working - but took a little on a Mac with VScode and Platformio.
I used these steps:

1.  Run  `ls /dev/cu.*` -> note the modem you see (e.g.: /dev/cu.usbmodem12345)

2. Short the B1 and GND pin, then press Reset (the blinking lamp near D4/D5 got solid for me). 

3. run  `ls /dev/cu.* `  again. Now a different modem is visible (e.g.: /dev/cu.usbmodem9C139E9136AC2)

Use that one!

4. Run this command:  `pio run -e esp32-s3-devkitc-1 -t upload --upload-port /dev/[YOUR FOUND MODEM HERE]` 

eg: `pio run -e esp32-s3-devkitc-1 -t upload --upload-port /dev/cu.usbmodem9C139E9136AC2`

Then it should work. 

5. Test in the pio console:  `pio device monitor -b 115200 -p /dev/[YOUR MODEM HERE]` 

eg: `pio device monitor -b 115200 -p /dev/[cu.usbmodem9C139E9136AC2`

I ran this config:

> [env:esp32-s3-devkitc-1]
 platform = espressif32
board = esp32-s3-devkitc-1
board_build.mcu = esp32s3
framework = arduino
build_flags =
  -std=c++11
  -DARDUINO_USB_CDC_ON_BOOT=1
  -DARDUINO_USB_DFU_ON_BOOT=1
monitor_speed = 115200
upload_speed = 115200
upload_protocol = esptool
lib_deps =
    locoduino/RingBuffer@1.0.5
    FrankBoesing/FastCRC@1.41.0
    PaulStoffregen/Time@1.6.1
    256dpi/MQTT@2.5.2
    adafruit/Adafruit BusIO@1.16.1
    adafruit/RTClib@2.1.4
    Wire@3.0.5
    SPI@3.0.5
    https://github.com/denyssene/SimpleKalmanFilter.git#v0.2
    ArduinoJSON@7.2.0

Hope it helps someone, as this comment helped me! Thank you @xtrmenl & @BmdOnline

PR #62 (2025-01-19): feat(platformio): add environment for esp32-s3-devkitc-1
Closes #56 

Build for arduino-esp32:

`pio run -e arduino_nano_esp32`

Build for esp32-devkit:

`pio run -e esp32-s3-devkitc-1`