tspopp/AquaMQTT #63: feature(wifimanager): get rid of configuration files

tspopp, 2025-01-19:
The community is having troubles in building AquaMQTT from source. Therefore...

- [ ] Add [WiFiManager](https://github.com/tzapu/WiFiManager.git) to configure AquaMQTT device after flashing
- [ ] Get rid of configuration file
- [ ] Provide pre-compiled flashable images of AquaMQTT
- [ ] Flash AquaMQTT images from web browser

BmdOnline, 2025-01-22:
Hi,

As a proof of concept, I'm trying to implement AquaDebug using EspHome.
In this way, we could configure wifi, MQTT, HomeAssistant automatically and dynamically.

First attempt looks promising :
[image]

I can read from UART and translate data into hex.
Now, I need to use MQTT component to send data automatically.

If it works, we may be able to try to convert AquaMQTT into an EspHome working version...
But I'm not sure I can do it alone.

BmdOnline, 2025-01-26:
Hi,
I have published my first working version of [AquaDebug ESPHome](https://github.com/BmdOnline/AquaMQTT/tree/main/tools/AquaDebug%20ESPHome).

I'm trying to port AquaMQTT now, but I'm not sure I'll be able to.

tspopp, 2025-01-26:
Nice! Any chance to provide some little docs on how to build this? I never used esp home. But sure, we can see if we can adapt the whole project.

ebbenberg, 2025-04-02:
I am also interested in an Esphome version.

It would be much easier to implement a custom ds18b20 or a i2c thermocouple k-meter iso on an Atom ESP32s3 lite

BmdOnline, 2025-04-02:
I'm still working on a prototype, but it's more of a proof of concept than a really usable version.
There are too many changes compared to the original code, and it will be difficult to maintain.
There's also the question of whether or not this Esphome version will support MQTT.

ebbenberg, 2025-04-03:
100% esphome without mqtt would be perfect for me.

I am not a programmer.

Some one transferred the MHI-AC-CTRL source code also to esphome
https://github.com/absalom-muc/MHI-AC-Ctrl/tree/master
https://github.com/ginkage/MHI-AC-Ctrl-ESPHome

I improved the V1 design a bit with SMD components and an M5stack Atom S3 lite

[image]