tspopp/AquaMQTT #64: About Atlantic Calypso

BmdOnline, 2025-01-19:
Hi,
I've managed to use ESP32S3 clone as explained here : https://github.com/tspopp/AquaMQTT/issues/56#issuecomment-2601013442

My heatpump use versions HMI F / Controller G.

I received a lot of `droppedBytes` and `msgCRCNOK`
[image]

So, I have installed AquaDebug and collected some data.
For now :
[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/18470339/aquadbg_debug_dec.csv)
[aquadbg_debug_hex.csv](https://github.com/user-attachments/files/18470340/aquadbg_debug_hex.csv)

If needed, I can collect more data.
Let me know.

Thanks.

tspopp, 2025-01-19:
This looks very good. It is the LEGACY protocol with a different checksum mechanism. Here are a few example messages:

```
194 35 33 2 4 0 2 0 0 0 4 88 32 48 0 0 0 0 15 51 18 20 21 0 0 0 0 3 0 0 0 0 0 255 255 255 153 
67 31 127 2 0 0 0 0 0 0 0 0 10 17 0 0 34 0 0 0 43 17 0 0 82 184 47 0 0 0 0 0 164 
193 37 194 1 139 1 184 1 0 0 90 0 0 0 0 0 0 0 6 0 0 45 0 64 255 255 255 0 83 71 38 2 0 0 14 1 65 0 101 
```

Seems they changed the checksum from two, to a single byte. Maybe it is the same checksum mechanism we have in the NEXT protocol.

BmdOnline, 2025-01-19:
It's a good new.
Let me know...

tspopp, 2025-01-19:
You might want to test my branch, I haven't tried this on my heatpump yet, but if I did nothing seriously wrong it should just work.

BmdOnline, 2025-01-19:
Yeah, it works :
[image]

BmdOnline, 2025-01-19:
A few minutes later...
[image]

Only one `msgCRCNOK`, some `droppedBytes`, but everything seems fine.
I will try MITM mode later.

All your changes (`platformio.ini`, `Configuration.h`) are great for `esp32-s3-devkitc-1` compatibility.

Thanks again.

tspopp, 2025-01-19:
Yeah, MITM is actually intresting, but I guess this will work out of the box, too. A few dropped bytes during initial protocol sync is actually expected but this value shouldn't rise further. The strange attributes which are shown as `{}` is actually an issue from MQTT Explorer and as far as I can tell not related to AquaMQTT. You won't have this issue in homeassistant.

closed 2025-01-21 by tspopp

svh227, 2026-10-11:
Hi everyone,

I am planning to transition my home to an all-electric setup and am highly considering purchasing the Atlantic Calypso Split V3 270L (the version that runs on R32 refrigerant). I want to integrate it locally into Home Assistant without relying on the cloud, which brought me to this amazing project.

Before buying, I have a few specific questions regarding this unit and its integration with AquaMQTT:

1. Hardware Check: Is the Calypso Split V3 270L (R32) the exact unit or one of the units you are currently using and testing this project with?

2. Temperature Limitation & Element Override: Out of the box, the factory software limits the water temperature setpoint to a maximum of 55°C, and it won't let you set it any higher on the controller. If I use the AquaMQTT board, can I successfully bypass this limit and force the boiler to go higher (e.g., up to 62°C) for PV-override/solar surplus matching?

3. Compressor vs. Element behavior: My understanding is that due to the physical properties and pressure limits of R32, the compressor itself simply cannot generate temperatures higher than 55°C without throwing errors. If AquaMQTT successfully overrides the target to >55°C, does the boiler automatically shut off the compressor at 55°C and engage the internal electric backup element (heating resistance) to bridge the remaining gap?

I would love to hear your experiences before pulling the trigger on this specific hardware combination.

Thanks in advance for your insights and for maintaining this great project!