tspopp/AquaMQTT #67: Atlantic Calypso VM vs Split Inverter

BmdOnline, 2025-01-27:
Hi,

We are discussing about Calypso compatibility [in this french thread](https://forum.hacf.fr/t/integration-dun-chauffe-eau-atlantic-explorer-en-mqtt/40493/195).

We are 4 users with these configurations :

| ESP32 | Heatpump | Version | Status |
| ------------- | ------------- | ------------- | ------------- |
| Official Arduino | Calypso VR | E/F | Failed |
| ESP32S3 clone | Calypso VR | E/F | Failed |
| Official Arduino | Calypso Split Inverter | F/G | Working |
| ESP32S3 clone | Calypso Split Inverter | F/G | Working |

After several tried, no data received, neither msgCRCKO ones.
AquaDebug grab no data either.
We are stuck !

Do you think changing port configuration may help ?
8N2 ? Baud rate ?

Do you have any leads ?

tspopp, 2025-01-27:
Well, no data is indeed strange. One of these wires must be a data line and you should be able to read something. Have these heatpumps three or four wires? Maybe the data is passed on the fourth wire? Are you are sure the gpio ports in the configuration file are correct? 

Initially I used this logic analyzer to see what is going on: https://www.az-delivery.de/en/products/saleae-logic-analyzer together with Logic 2 to get the first idea of how the protocol is working. Maybe you are right, and the split inverter is using some kind of different protocol. But still I would expect, that the Arduino will read something on serial data line, even if it is garbage.

BmdOnline, 2025-01-27:
Both heat-pumps uses 3 wires.
One of the members is using the same ESP32S3 as me, with the same GPIO configuration.

I have used the same analyzer for debug during my firsts attempts.
But I dont think others members have this. I don't know what to advise them.

tspopp, 2025-01-27:
Strange. Mh, few more thoughts:

- Make sure Arduino and heatpump share the same GND, if GND is not connected you might read nothing as well
- Make sure the communication between HMI and Controller is good (HMI gets temps from controller, HMI is able to select modes, e.g. BOOST)
- Try tapping the data line (see above) while HMI and Controller are communicating.

tspopp, 2025-01-27:
Ah, it seems officially the esp32 is officially not 5V tolerant, so it is risky to do this. Forget my comment 😬

sebcaps, 2025-01-27:
Hello, 
One of the guy having trouble with reading data. I confirm you that when aquamqtt is plugged hmi is ok ( data shown on the screen) but did not try to send command (eg boost)
I'll have a look to logic analyser. Don't have one yet but why not give a try 😉

BmdOnline, 2025-01-28:
@tspopp,
What do you think about using an USB-FTDI adapter instead of Arduino, but without connecting VCC.
Connecting only GND and RX pins, and check if data are received ?

tspopp, 2025-01-28:
This should work as well. But if you need to debug the serial protocol (baud rate, etc) the logic analyzer is very helpful.

BmdOnline, 2025-01-28:
> the logic analyzer is very helpful.

@sebcaps, _y'a plus qu'à_ \* 😉 

\* Let's do it

sebcaps, 2025-02-01:
Hello,
 I gave a try to logic analyser and except on pin D5 I can't see anything.
With default serial settings value correspond to 0XC2.

[image]

Other info : while Aquamqtt is plugged display is OK BUT any commands (boost, fan test) does not trigger anything (no bip for boost, no fan start / stop...)
If it may help the recorder SAL file(jsut remove txt extension)

[SessionD5-Chan4 - Copie.sal.txt](https://github.com/user-attachments/files/18628290/SessionD5-Chan4.-.Copie.sal.txt)

tspopp, 2025-02-01:
> Other info : while Aquamqtt is plugged display is OK BUT any commands (boost, fan test) does not trigger anything (no bip for boost, no fan start / stop...)

Are you in Listener Mode? 🤔

sebcaps, 2025-02-02:
Some (good) news; after a day by 5°:
- I had a connectivity issue (using adapter to not cut wire) : I had a short between GND and data line in my connector : fixed (tackled connecting HMI & main board and sniffing data (ie no PCB no arduino)
- While using PCB (even in LISTENER MODE and J7 short) => no data,
- Connecting 5V, GND and data line from main to Arduino and to HMI according to the following schema I got data (DroppedBytes only) 

[image]

- Ran Debug and got Hex data . I 'll need to install debug tool . For the moment too cold to do so ;-)

Another user with the same model (Calypso E/F)) is also reporting issue reading data using transistor and level shifter. Ie no data, but OK when plugged directly....

tspopp, 2025-02-02:
It is indeed strange that the logic-level-converter (LLC) does not work for those machines. Maybe these machines already provide 3.3V logic level? You might want to check with a multimeter? You may also check, if the cable which is expected to provide +5V, if that is really providing +5V or something else.... 🤔 🤷‍♂ 

Happy you have finally read something from the heatpump. If AquaMQTT is not able to parse the messages already, just drop a dump (dec or hex) from AquaDebug and we determine the cause.

Arrnooo, 2025-02-02:
Hello,

I suspect that the serial link is very sensitive to impedance changes.
This would explain the communication issues as soon as a component with a pull-up resistor is connected to the link, whether in listening mode or MITM.

When directly connected, with the data wire attached to GPIO_MAIN_RX, all information is received. This rules out a software issue.

I took measurements with an oscilloscope. The voltage is indeed at 5V.

And with two logic buffers like the 74LVC2T45?

tspopp, 2025-02-03:
Nice analysis! What is the path forward? Will you proof a new hardware design which resolves the issue for these heat pumps? I

I can help verifying if the new hw design will also work with my legacy heat pump. And of course I can incorporate changes with KiCad and put a rev.2 there.

Arrnooo, 2025-02-03:
[image]

I would like to test this version of the assembly.

However, the buffers are bidirectionally controlled.
High = A to B
Low = B to A

This requires adding a TX enable in the software, set to high only during the transmission phase.

Do you think this is possible for you?

tspopp, 2025-02-03:
I put it on my branch here: https://github.com/tspopp/AquaMQTT/tree/t/hwpoc

You may adapt `Configuration.h` to set the correct `ENABLE_TX_HMI` and `ENABLE_TX_MAIN` GPIOs accordingly. Those GPIOs will be set to high whenever AquaMQTT is sending something on a dedicated channel. In Listener mode they are of course unused.

Arrnooo, 2025-02-03:
Super!

I ordered the components. I'll engrave the PCB with my CNC in the meantime. As soon as everything is ready, I'll test it and give you feedback.

BmdOnline, 2025-02-03:
Hi,
what about 74LVC2T45 vs logic inverter currently in use ?
What are its advantages ?

If you're working on a rev.2, maybe you can add a way to disable the entire board (power and data) using a rocker switch.
I plan to adapt an intermediate board connected to HMI and Main ports J1 & J2, allowing to disable everything, remove ESP if needed, only using a rocker switch without powering off the heatpump.

tspopp, 2025-02-03:
As far as I understood it mainly resolves the need for the LLC board and the transitors. So we replace five parts by two. Moreover it resolves the issue with those heatpumps, where we are not able to read the serial bus due to impedance changes (due to the LLC).

Arrnooo, 2025-02-04:
Ok.
If I analyze this correctly, these are the messages sent by the main board.
Since the connection is disrupted, the HMI receives nothing and therefore does not respond. As a result, the main board keeps sending the same three messages in a loop.

This supports my hypothesis of a connection disrupted by the addition of an element that listens to the communication.

tspopp, 2025-02-04:
Hm, something is wrong in those traces, since it is an endless combination of message identifiers without data:

It is always the same pattern `194 67 193` which is repeated. Usually after each message identifier, there is a message length field and some data. In this case there is nothing, so there is no data in those traces. I am wondering what went wrong here...  🤷‍♂

tspopp, 2025-02-06:
@Arrnooo let me know if the new actually design works. In the meantime I prepared a new PCB. This time we should be able to order it pre-assembled by JLCPCB and it provides the RAST adapter designed by @tukutt as well.

[image]

BmdOnline, 2025-02-07:
Hi,
Are you sure about DIP format for U1 and U2 ?
According to [SN74LVC2T45 datasheet](https://www.ti.com/product/SN74LVC2T45#pps), it is only provided as surface mount component.

tspopp, 2025-02-07:
You are right, this is an early draft. I've been just too excited yesterday. Will finish this as soon as there is feedback if that actually works.

BmdOnline, 2025-02-07:
@Arrnooo,
is there an alternative to SN74LVC2T45, but in DIP format ?
It might be interesting to avoir surface mount soldering.

Arrnooo, 2025-02-07:
The ordered components are surface-mount components. However, my CNC is not precise enough to engrave such small traces. So I also ordered adapters for this type of DIP footprint. A choice will have to be made for the final PCB.

I will receive the components this week and engrave the PCB this weekend. Expect an update in a week if time permits.

Arrnooo, 2025-02-12:
I made a mistake reading the datasheet. The component is far too small for me (0.5mm pitch instead of 0.65mm).

I finally found an equivalent in a through-hole component (SN74LS245). I'm going to test it with just one. Do you think this will cause errors?

It's on order. I'll keep you updated.

[image]

tspopp, 2025-02-13:
I don't think this will work: In MITM Mode, there are two individual serial sessions going on. In case the controller session wants to read and the hmi session wants to write, you are in big trouble and you will loose packets🤔 

If we get the board manufactured and pre-soldered, we don't really have an issue with using the small components, right?

Arrnooo, 2025-02-23:
It’s working with the SN74LS245 in listening mode. I took measurements with the oscilloscope, and everything looks good. So, I believe the solution with the two 74LVC2T45 should work as well.

If you have the PCBs manufactured and the components soldered, it should be feasible.

sebcaps, 2025-02-23:
Thx @Arrnooo  for your work and feedback
@tspopp do you think you'll manage to setup a new PCB design based on Arrnooo schematic (for an arduino)  ? I could give a try but not a pcb designer expert :-). Definitely I don't uderstand the component change nor H1 /H4 purpose (both are tied to the same   net). SN74LS245 exist in  through hole, so I guess this is a good 'alternative'

tspopp, 2025-02-23:
Yeah, I will take care 👍 Love to get a few things optimized such as RAST and pre-soldered through jlpcb.

tspopp, 2025-02-24:
In case I did not make any mistakes, you can find a working revision 2.0 of the board [in my branch](https://github.com/tspopp/AquaMQTT/tree/t/hwpoc/pcb). It is meant to be bought pre-assembled by JLCPCB.

While I guess that everything will be good, I am bit torn If I should invest another 30$ (5 boards) for purely academic purposes. My heatpump is running very well with rev1.0 and rev2.0 does not make any differences to me. Moreover, I currently really don't want to invest time in de-assembly my heatpump one-more-time and switch boards 😬 But I am still considering this.... 🤔 

Maybe @Arrnooo might willing to review the schematics, but I guess it is good. Actually I would order it right away ;)

[image]

Arrnooo, 2025-02-25:
Everything seems to be in order.

I don't think I'll use your PCB. I'm planning to make my own version with a connector for controlling a relay and another for a PWM signal, with the aim of varying the power of the water heater's heating element and utilizing the surplus production from my solar panels.

Thanks for all your modifications and help.

Arrnooo, 2025-03-08:
Everything is functional with my through-hole equivalent component.
The communication direction seems to be well managed.
I think the "pinMode" settings for "enable TX" were missing, which I have now added.

The information is properly transmitted from the water heater to the MQTT server, and the target temperature modifications (among other things) from the MQTT server to the water heater.

However, I have about 100 "dropped bytes" per second on the HMI. Is this normal?
On the Main, it's close to 0.

tspopp, 2025-03-08:
No, constant dropping is an issue. Let me check if the "enable TX" is correctly used. I guess you already checked that the HMI controller is not sending an additional message next to 194?

Arrnooo, 2025-03-08:
No, I did indeed use two SN74LS24—one for the Main and one for the HMI.

Is it normally possible to activate the resistor in test mode from the HMI?
For me, it doesn’t work when the board is in MITM mode.

tspopp, 2025-03-10:
Yeah, in MITM mode you should be able to control the heatpump from the HMI just like you did before. I always test If I am able to run the heatpump in BOOST and get temperature values shown in the HMI. An additional test is, to open the super secret menu and see if you are able to receive stored error messages.

I have a few more ideas why there is dropping:

- The heatpump HMI is sending additional data: You may install AquaDebug in Listener Mode and record a trace of the communication and we identify the raw messages in trace, inspecting if there is something new.
- Something is blocking the hmi task due to workload. You mentioned to add some code for a PWM signal. Are you still using the original version of my code or is it already altered?
- Actually, I rewrite portions of the MITM mode for supporting the fairly old atlantic odyssee heatpump in #68. As soon as I pushed my changes, we may want to test with this version and see if the drops are still there.
- At very last I can also add some code, to determine which bytes are being dropped. With that, we might identify the reason for the dropping.

Arrnooo, 2025-03-12:
No, I'm still using the dedicated branch.

Here are the extracted frames.

[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/19208752/aquadbg_debug_dec.csv)
[aquadbg_debug_hex.csv](https://github.com/user-attachments/files/19208753/aquadbg_debug_hex.csv)

I might want to try extracting the actual sent/received frames in MITM mode with a logic analyzer.

Quentin90000, 2025-03-13:
Hi, it's great how you've made progress on the subject, you have something concrete! As soon as it works I can get back to soldering work 😅.
Arrnooo, do you have a Calypso in E/F version like Seb and myself?

tspopp, 2025-03-19:
Yes, I provided [a fix](https://github.com/tspopp/AquaMQTT/commit/2fa5a54ce1bf603009104a55afa3cbb0841a1cdf) to the odyssee branch. 

Background: With the rev1.0 board there is always a loopback, so what is sent on the wire is also received. With the rev2.0 hardware there is no loopback anymore. In this case, the hmi message identifier used for requesting the hmi message from the hmi controller, was never written into the message buffer. Therefore the whole buffer contained a complete hmi message, but the identifier in the beginning has been always missing. That's why the whole thing got dropped.

Arrnooo, 2025-03-21:
Ok.

I applied the fix. Indeed, it resolved many issues. Now I can run tests from the HMI. That’s great!

I still have some dropped bytes. I’ll share them with you.

[aquamqtt_stats_hmi_droppedBuffer_dec.csv](https://github.com/user-attachments/files/19388842/aquamqtt_stats_hmi_droppedBuffer_dec.csv)

[aquamqtt_stats_hmi_droppedBuffer_hex.csv](https://github.com/user-attachments/files/19388843/aquamqtt_stats_hmi_droppedBuffer_hex.csv)

tspopp, 2025-03-21:
There is still an race condition: It seems sometimes we are not fast enough setting the enable pin to LOW and therefore damaging the message. Since these messages are replayed many times, the current implementation is still working.

Actually, I ordered a set of the rev2.0 boards. I will fix this issue as soon as they are delivered.

tspopp, 2025-03-27:
I got my set of rev2.0 boards and it works as it should. As expected, there was still a race condition. I made a fix in https://github.com/tspopp/AquaMQTT/pull/69/commits/c5286657b823d33e32b93196404c48fff5e0a9f4 on the odyssee branch. I will soon prepare my PRs and get this merged.

closed 2025-03-27 by tspopp

tspopp, 2025-03-27:
I closed this, since I believe the problem has been solved with the new circuit and hardware. You may just order new pre-assembled pcbs through JLCPCB, see [my instructions](https://github.com/tspopp/AquaMQTT/tree/main/pcb) and provide feedback.

sebcaps, 2025-03-30:
> I got my set of rev2.0 boards and it works as it should. As expected, there was still a race condition. I made a fix in [c528665](https://github.com/tspopp/AquaMQTT/commit/c5286657b823d33e32b93196404c48fff5e0a9f4) on the odyssee branch. I will soon prepare my PRs and get this merged.

Hello @tspopp  do you use all the boards ? I guess 5 were manufactured (minimum on JLCPCB). If you have extra one(s), would you agree to sell them ? I'm interested in having one and sure at least 2 other will be.

tspopp, 2025-03-30:
Hey @sebcaps, I have three left, since I already sold one. I added a discord link to the main page. Let's discuss there.

sebcaps, 2025-04-27:
Hello there.
Finally I ordered rev2 PCB directly (definitely dirty cheap...) and.... it works. 
Definitely v2 of the board is compatible with Atlantic Calypso version E/F with an arduino clone !
As we say in french "y'a plus qu'a" .