tspopp/AquaMQTT #68: Atlantic Odyssée Pi Debug

FSlunkk, 2025-02-09:
Hy,
I tried to connect the aquamqtt to my atlantic odyssee pi.
I have data arriving but it is not recognized.
Please find attached the data collected. 
Is there anything we can do to make this work?

[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/18725290/aquadbg_debug_dec.csv)

tspopp, 2025-02-09:
Hi @FSlunkk,

this looks like another protocol which we haven't seen so far.  Your dump appears to have four different messages:

`194 29 54 2 0 33 0 0 0 0 0 0 0 0 0 0 0 0 33 50 38 17 0 0 0 0 65 66 0 0 47`
`67 25 64 0 251 15 0 0 0 0 195 0 0 0 176 46 0 0 0 0 0 0 0 0 0 0 240`
`217 31 31 0 0 232 0 1 0 129 1 236 210 88 0 14 1 204 0 0 1 0 0 2 0 0 0 0 0 0 0 0 0 208`
`193 29 29 53 0 31 0 14 0 14 0 18 0 0 0 53 0 47 0 0 0 0 0 0 2 0 0 0 0 82 66 4`

So they seem to use the same identifier `194`, `193` and `67` which we also observed within both other protocols but the message length differs. And there is a new message `217` which Is completely new.

I can add parsing of these new messages to AquaMQTT on a seperate branch, but it would be your work in identifying values within the data, very similar to what we did in https://github.com/tspopp/AquaMQTT/pull/52 

I guess that this protocol will share a lot of similarities with the existing protocols [NEXT](https://github.com/tspopp/AquaMQTT/blob/main/PROTOCOL_NEXT.md) and [LEGACY](https://github.com/tspopp/AquaMQTT/blob/main/PROTOCOL.md), so it might be a good read as well.

What kind of software version does your Controller and HMI unit show? Also see https://github.com/tspopp/AquaMQTT/issues/45

FSlunkk, 2025-02-09:
[image]
I will look at #52 and try to do the same work.

tspopp, 2025-02-09:
Ok, nice, then I'll prepare something for you to start :+1:

FSlunkk, 2025-02-10:
OK thanks. I'll try to start this week when I have some free time.

tspopp, 2025-02-10:
@FSlunkk you may proceed in #69 :) I would be very happy if you edit the [PROTOCOL_ODYSSEE.md](https://github.com/tspopp/AquaMQTT/blob/t/odyssee/PROTOCOL_ODYSSEE.md) file directly and provide your findings directly to the document (fork and create PR). Looking forward!

FSlunkk, 2025-02-10:
Thanks, i started to collect some data, I will analyze it during the week.

closed 2025-03-27 by tspopp