tspopp/AquaMQTT #74: Using element only on Atlantic V3

eterpstra, 2025-04-07:
Hi,

I have installed the AquaMQTT module.  Everything seems to work, but for some reasons i am not able to use the element only modus. I have setup the following as mentioned in the FAQ:
[https://tweakers.net/fotoalbum/image/PNsoNaETxv1jHAUhK6jbrbxD.png](url)

Am i doing something wrong? Is there anyone that is using a V3 Atlantic Explorer with the same problems? The AquaMQTT heating element stays ''OFF"

tspopp, 2025-04-07:
Interesting, can you confirm your heat pump is using the NEXT protocol? It is provided as attribute to via MQTT to `aquamqtt/stats/protocolVersion`. You may inspect this value using MQTT Explorer.

Have you tried forcing the heat element usage without AquaMQTT? The PV Mode HeatElement is just a shortcut to enable several overrides at once, see also the [PV README](https://github.com/tspopp/AquaMQTT/blob/main/README-PV.md#pv-heat-element-only-pv-he). In theory, if you set the following attributes within the original HMI Controller, your heat pump should start heating up with the heating element only:

- Installation Mode is set to HEATPUMP ONLY (we don't want to use an external boiler if any connected)
- OperationMode is set to "MAN ECO OFF"
- OperationType is set to "ALWAYS ON"
- Water Temperature is set to 62°C
- Heating Element is set to enabled
- Emergency Mode is turned on (forces usage of heat element)

Another note, the HMI Controller is just telling the Main Controller a desired state. The main controller might still decide to not heat up, for example, the water temperature reached already its threshold and the heatpump controller decides not to turn on.

tspopp, 2025-04-07:
Yeah, that might be the case. Good find. Can you do me a favor and proof this? You need change the Configuration file and reflash:

```
/**
 * Choose to publish raw messages represented as hex-string on debug mqtt topics
 */
constexpr bool DEBUG_RAW_SERIAL_MESSAGES = true;
```

You will then see debug messages on topic `aquamqtt/hmi/debug`. 

Then, in the original HMI controller (not aquamqtt), set the "allow heating element" switch to ON and copy paste the debug message here.

Then, again, set it to off in the original HMI controller and copy paste the debug message here.

eterpstra, 2025-04-07:
hmmm have to look up for that because my device was pre-flashed. Is it not possible via OTA?

tspopp, 2025-04-07:
Yes, you may flash via OTA!

I prepared a possible [fix](https://github.com/tspopp/AquaMQTT/commit/0ae6a9ca774aa09edf97dd9588a02085b5b4f7cd) in my [branch](https://github.com/tspopp/AquaMQTT/tree/t/next_fix_pvhe). You may also directly try if this works for you. 

Protocol dumps of your heatpump would be still very good, since I like to proof that this is really the issue. I will await your feedback ;)

eterpstra, 2025-04-07:
Thank you i have to look into that. It is not possible to flash it over the air (Like ESPhome)? I am on a bussiness laptop and that does not allow to install any software hehe.

tspopp, 2025-04-07:
Have a look in the readme within the AquaMQTT folder. Unfortunately it is not as easy as ESPHome. You need at least platformio and execute a single command...

eterpstra, 2025-04-07:
A friend of my tried, and this is the output:

`src/message/next/HMIMessage.cpp:7:1: error: no declaration matches 'aquamqtt::message::next::HMIMessage::HMIMessage(uint8_t*, const uint8_t*)'
    7 | HMIMessage::HMIMessage(uint8_t* data, const uint8_t* previous) : mData(data)
      | ^~~~
In file included from src/message/next/HMIMessage.cpp:1:
include/message/next/HMIMessage.h:11:7: note: candidates are: 'aquamqtt::message::next::HMIMessage::HMIMessage(const aquamqtt::message::next::HMIMessage&)'
   11 | class HMIMessage : public IHMIMessage
      |       ^~~~
include/message/next/HMIMessage.h:14:14: note:                 'aquamqtt::message::next::HMIMessage::HMIMessage(uint8_t*, uint8_t*)'
   14 |     explicit HMIMessage(uint8_t* data, uint8_t* previous = nullptr);
      |              ^~~~
include/message/next/HMIMessage.h:11:7: note: 'class aquamqtt::message::next::HMIMessage' defined here
   11 | class HMIMessage : public IHMIMessage
      |       ^~~~
src/message/next/HMIMessage.cpp:13:6: error: no declaration matches 'void aquamqtt::message::next::HMIMessage::compareWith(const uint8_t*)'
   13 | void HMIMessage::compareWith(const uint8_t* data)
      |      ^~~~
include/message/next/HMIMessage.h:46:10: note: candidate is: 'void aquamqtt::message::next::HMIMessage::compareWith(uint8_t*)'
   46 |     void compareWith(uint8_t* data);
      |          ^~~~~
include/message/next/HMIMessage.h:11:7: note: 'class aquamqtt::message::next::HMIMessage' defined here
   11 | class HMIMessage : public IHMIMessage
      |       ^~~~
src/message/next/HMIMessage.cpp: In member function 'virtual uint8_t aquamqtt::message::next::HMIMessage::getAttr(aquamqtt::message::HMI_ATTR_U8)':
src/message/next/HMIMessage.cpp:185:24: error: 'OT_ALWAYS_ON' was not declared in this scope; did you mean 'ALWAYS_ON'? 
  185 |                 return OT_ALWAYS_ON;
      |                        ^~~~~~
      |                        ALWAYS_ON
src/message/next/HMIMessage.cpp:187:20: error: 'OT_TIMER' was not declared in this scope; did you mean 'TIMER'?
  187 |             return OT_TIMER;
      |                    ^~~~
      |                    TIMER
src/message/next/HMIMessage.cpp: In member function 'virtual void aquamqtt::message::next::HMIMessage::setAttr(aquamqtt::message::HMI_ATTR_U8, uint8_t)':
src/message/next/HMIMessage.cpp:386:34: error: 'OT_TIMER' was not declared in this scope; did you mean 'TIMER'?
  386 |             if (operationType == OT_TIMER)
      |                                  ^~~~
      |                                  TIMER
src/message/next/HMIMessage.cpp:390:39: error: 'OT_ALWAYS_ON' was not declared in this scope; did you mean 'ALWAYS_ON'? 
  390 |             else if (operationType == OT_ALWAYS_ON)
      |                                       ^~~~~~
      |                                       ALWAYS_ON
* [.pio\build\esp32-s3-devkitc-1\src\message\next\HMIMessage.cpp.o] Error 1`

tspopp, 2025-04-07:
Hm, [my CI passes sucessfully](https://github.com/tspopp/AquaMQTT/actions/runs/14315915628/job/40121845071) so issue must be on your end

It seems like you don't have a clean branch, since enum types have been renamed recentely. I suggest solving with a fresh `git clone`:)

eterpstra, 2025-04-07:
I have installed Visual Code server as an addon now in HA. But i am not able to find platformio. Is that correct?

eterpstra, 2025-04-08:
@tspopp Hi,

I am going to install a VM with ubuntu. Ive read the manual, and used  git clone https://github.com/tspopp/AquaMQTT.git

However, i dont see a ''config'' directory within the cloned project. Is this normal?

eterpstra, 2025-04-08:
So @tspopp  I a friend of my has installed the code on a v4 (before you made the change in de code) and then the element is working fine.

So it is related to the V3 then i quess. Unfortunately i am not able to test the new code yet.

aeonfast, 2025-04-08:
I installed the AquaMQTT board today as well, same on as TS on a v3 and I'm experiencing the same issue that I cannot get the electric element working. Will keep an eye on this thread.

eterpstra, 2025-04-09:
@tspopp I tested the branch with the fix, unfortunately it seems it now doenst do anything anymore.

tspopp, 2025-04-09:
Yeah, please provide dumps. A single message when "enable heat element" was OFF and a single message when "enable heat element" was ON. 

Don't get confused by the tons of messages, they are changing a lot because it contains other data as well (current time for example).

eterpstra, 2025-04-09:
But how do i know what the message is for enable heat element OFF and enable heat element ON?

eterpstra, 2025-04-09:
> Set it in the original hmi controller?

Yes.

What i can see with MQTT Explorer: When i enable heatelement i see the following change:

StatePV = 1

SHould that not be StateElement = 1 ?

eterpstra, 2025-04-09:
> [Discord](https://discord.gg/dYV2CPGs)?

Sure, but link is expired.

tspopp, 2025-04-09:
@taloriko: Maybe you can jump in and validate if the [fix](https://github.com/tspopp/AquaMQTT/tree/t/next_fix_pvhe) for the V3 / NEXT protocol is also good for your machine? Actually it seems that the "allow/enable use of heat element" setting from the engineering menu has been interpreted in the wrong way. So if the setting was set to "allowed" in the HMI controller, AquaMQTT showed it as "disabled".

taloriko, 2025-04-09:
Sure, I'll see if I can flash the fix tomorrow at noon.

I haven't worked with the heating element so far,
but I plan to integrate it into my energy management anyway.

tspopp, 2025-04-09:
Thanks @taloriko! Actually @eterpstra was able to verify the fix moments ago. It seems to be good :)

It also matches the protocol findings you did back then. So I am happy to get this fix merged soon.

closed 2025-04-09 by tspopp

taloriko, 2025-04-13:
Sorry for the late message.

Today, I installed the current version 1.7.0. Was that the correct one, or should I be using a different version?

When I changed the HMI of the heating element, I expected the result to appear under the topic aquamqtt/hmi/heating_element, but there was no change.

I  have the following options on the HMI:

[image]

[image]

taloriko, 2025-04-13:
False alarm – it works after a restart:

[image]

[image]

I suspect the problem lies elsewhere.
Back then, I defined the byte like this:
[image]

The byte behaves differently depending on whether PV-Grid is set to 'NO' or 'YES' in the hidden menu.
When PV-Grid is 'YES', 2 is added.

[image]

[image]

PV-Grid "NO"
Element OFF = 4
Element AUTO = 0

Bei PV "YES"
Element OFF = 6
Element AUTO = 2

taloriko, 2025-04-14:
Okay, Git still showed version 1.7.0. In the version.h I tested with, it already says 1.7.1.

With version 1.7.1, the heating element works correctly.

I also integrated it directly into my energy management system.

The functions look good there too.
The heat pump and heating element can be controlled correctly.

[image]

taloriko, 2026-01-02:
Hi Phil-Scheider,
everything is well described here, and I implemented it exactly like that:

https://github.com/tspopp/AquaMQTT/blob/main/WIRING.md

Arduino + PCB Rev 1 works fine for me.
I still have some PCB Rev 1 boards left. If you like, I can send you one.

phil-schneider, 2026-01-02:
Thank you very much for your response.
Today, I just ordered v2.
Looking forward to finally Controll my water heater.
Currently just using the pv input with a shelly.

PR #75 (2025-04-09): fix(next): fix option allow heat element and pv mode heat element
Fixes #74