tspopp/AquaMQTT #80: Interface graphique E & régulation H

zorly57, 2025-05-02:
Hello,

I'm using the E graphical interface and H control. In MIMT mode, nothing works, and in LISTENER mode, only the programmed temperature and a few other elements are displayed, including the attachment.

(Sorry, my English is very poor.)

Sincerely,

[image]

[image]

[image]

[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/20018135/aquadbg_debug_dec.csv)
[aquadbg_debug_hex.csv](https://github.com/user-attachments/files/20018136/aquadbg_debug_hex.csv)

tspopp, 2025-05-03:
Well this leaves me some riddles. Your dump contains three message types: 

```
C2 23 1C02323C10000000005820301000000023A212111600000000080000000000FFFFFF F183
43 1F 59431E00C5E22D04001000E91000C5080000D7F9
C1 25 17B35B5B000000000A400102FF57086102000E5436FE
```
---
There is the HMI message, sent from the HMI controller with identifier `0xC2` -> `194 dec` with message length `0x23` ->` 31 bytes` which matches the [LEGACY](https://github.com/tspopp/AquaMQTT/blob/main/PROTOCOL.md) protocol and is parsed correctly. These are the values you see in HomeAssistant / MQTT. The last two bytes `F183` is the checksum which is also CRC-16/CCITT-FALSE which is good ✅ 

----

Then we have the main and and the energy message emitted by the controller. These look incomplete:

Identifier `0x43` ->` 67dec`, expected message length `0x1f` -> 31 bytes. This message is therefore not complete, or there is some weird new interpretation of the length field. If I treat the last two bytes as checksum, I did not found a matching checksum algorithm either. So my current expectation is, that for some unknown reason the messages sent from the main controller are not complete.

Same applies to the main message `0x1c` -> `193dec`: Length `0x25` (37 bytes) are expected, message is incomplete.

---

So at this point I don't really have answers.

- Is your heatpump working in LISTENER mode as expected? Most important, does your HMI controller show current temperature values from the boiler?
- I would most probably use an logic analyzer to check if the messages are complete if AquaMQTT is not connnected at all.

zorly57, 2025-05-03:
Thanks for the feedback, in LISTENER mode the heat pump works perfectly and I can see the values ​​on the screen in installer mode:

[image]

[image]

[image]

[image]

In MITM mode the heat pump no longer works:

[image]

[image]

[image]

I don't know much about programming, but when I open the aquabg_debug_hex file, it seems to be correct, but as I said, I don't know anything about it.

[image]

I ordered a logic analyzer which should arrive tomorrow, I hope it's good: 
https://www.amazon.fr/dp/B07D21GG6J?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

tspopp, 2025-05-04:
Yes, the dump looks fine, but the messages from the controller are incomplete. I don't know what's the reason for this.

zorly57, 2025-05-04:
Here is a capture almost simultaneously with the aquamqtt debug and with the logic analyzer.

[50s logic analyzer.csv](https://github.com/user-attachments/files/20030254/50s.logic.analyzer.csv)

[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/20030256/aquadbg_debug_dec.csv)
[aquadbg_debug_hex.csv](https://github.com/user-attachments/files/20030255/aquadbg_debug_hex.csv)

tspopp, 2025-05-05:
Looking at the dumps, the logic analyzer was able to record the whole messages:

```
id len   data                                                                       checksum
C1 25    DE00BC009C00A2000000000000000000000000000000FFFFFF57084858020000C8005400   5CF1
C2 23    1C02323C10000000005820301000000017A512040000000000080000000000FFFFFF       C2FE
43 1F    35C0C81E0000FDDEE2AFEA04000004000000EA040000FDDE080000000000               279F
```

while AquaDebug was only able to record portions of these messages

```
C1 22    517B35B5B000000000A400102FF57086102000E5436FE 
C2 23    1C02323C10000000005820301000000023A212111600000000080000000000FFFFFF       F183
43 1F    59431E00C5E22D04001000E91000C5080000D7F9 
```

I don't expect any sw issues in AquaDebug since it is just reading from Serial and providing dumps to MQTT. So yes, maybe this is related to the v1 pcb or the components soldered on the v1 pcb. I am not sure, I've never seen this issue before.

The good news is, that this protocol is supported by AquaMQTT. If you can resolve the issue, then it should work as expected...

zorly57, 2025-05-05:
I tested other resistance even if the value was good before because I had others but it didn't change anything. There are still transistors, the logic level converter and the arduino, I will recommend these elements because I think there is no easy way to know where it comes from. I also have a pcb that did not work at home on the way to a heat pump with a previous Main firmware.

zorly57, 2025-05-05:
Ok, I'll order it later in this case. Thanks again for the feedback. I keep you informed of this. In case my water driver is in HMI = E and MAIN = H version, it was purchased in France in March 2024.

zorly57, 2025-05-22:
Hello, I just received V2 today and there are no changes compared to V1.

[image]

[image]

[image]

zorly57, 2025-05-22:
Voici les dump aquadebug :

[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/20390377/aquadbg_debug_dec.csv)
[aquadbg_debug_hex.csv](https://github.com/user-attachments/files/20390376/aquadbg_debug_hex.csv)

tspopp, 2025-05-22:
Well, you are right. Nothing has changed messages are still incomplete. I am not sure, why it isn't working with your heatpump, but I am pretty sure we will find the reason.

In my [branch](https://github.com/tspopp/AquaMQTT/tree/t/buffer) I increased internal buffer sizes of AquaDebug. Let's try these changes and create a new message dump. Hopefully, we will see complete messages.

zorly57, 2025-05-23:
Hello, here is the dumps with the new buffer. 
thanks for help!

[aquadbg_debug_hex.csv](https://github.com/user-attachments/files/20407061/aquadbg_debug_hex.csv)
[aquadbg_debug_dec.csv](https://github.com/user-attachments/files/20407060/aquadbg_debug_dec.csv)

tspopp, 2025-05-23:
Unfortunately, this was not reason, they are still incomplete. Do you have another recoding from your logic analyzer, showing raw data, not already interpreted data? Currently AquaDebug expects 8 bits data and 2 stop bits  (SERIAL_8N2) at an baud rate of 9550:

```
// setup serial port
Serial2.begin(9550, SERIAL_8N2, GPIO_MAIN_RX, GPIO_MAIN_TX);
```
[main.cpp line 110  ](https://github.com/tspopp/AquaMQTT/blob/0aa134f01930be2fa454b3bda708686c8bf3e6c7/tools/AquaDebug/src/main.cpp#L110)

Maybe you can play with different settings. I would try a baud rate of `9600` instead of `9550` because it is awkward anyway, but was the most fitting baudrate according to my initial logic analyzer traces. I never changed that, since it just worked. But I guess `9600` does make more sense.

Maybe also try with different stop bits, such as [SERIAL_8N1](https://docs.arduino.cc/language-reference/en/functions/communication/serial/begin/).

Have you also tried another Arduino? Just in case something is broken on that one?

tspopp, 2025-05-23:
Yes, now it looks perfect. All messages are complete 👍 You can also try MITM if you change this setting [HMITask.cpp](https://github.com/tspopp/AquaMQTT/blob/0aa134f01930be2fa454b3bda708686c8bf3e6c7/AquaMQTT/src/task/HMITask.cpp#L49) and in [ControllerTask.cpp](https://github.com/tspopp/AquaMQTT/blob/0aa134f01930be2fa454b3bda708686c8bf3e6c7/AquaMQTT/src/task/ControllerTask.cpp#L44)

What values are missing? You might also refer to the [MQTT documentation,](https://github.com/tspopp/AquaMQTT/blob/main/MQTT.md) to see what is currently supported with the LEGACY protocol :)

Very happy we found the issue, I will move this setting to the Configuration file soom, so it can be easily adapted for heatpumps which need SERIAL_8N1 instead of SERIAL_8N2.

I am a bit sorry that in the beginning we suspected the hardware to be the actual issue. But I guess you will be able to sell the v2 PCBs very easy. I guess there will be people on the HACF forum looking for v2 boards :)

zorly57, 2025-05-23:
I just tested V1. It seems to work only with the HMI, but nothing is received on MAIN, whereas it is with V2.

I'll test this on MITM and keep you updated on that part.

Currently, in LISTENER mode, when I restart HA, certain data no longer appears. I need to restart the ESP nano. The temperature also increases slowly. For example, "Water Temperature" is displayed when the ESP nano starts: 29.1°C, then 40.6°C, then 50.1°C, then 50.3°C (actual value).

zorly57, 2025-05-23:
Here is a screenshot of MQTT Explorer :

[image]

here is the DEBUG line of the energy part: 1FA5E7C81E00006D063DB10105000004000000010500006D06090000000000

I would like to point out that my heating element is deactivated and never heats up (4 hours since purchase)

My pump has finished heating up. I'll run the fan tests tomorrow, but in the meantime, here's what I saw in the operator menu:

Fan on 1:
Fan PWM = 0,0%

[image]

[image]

Fan on 2:
Fan PWM = 0,1%

[image]

[image]

tspopp, 2025-05-24:
Can you provide a single dump of the main message (aquamqtt/main/debug) when...
a.) fan running in speed 1
b.) fan running in speed 2
c.) fan turned off

tspopp, 2025-06-10:
@zorly57 Is it possible for you to provide these traces so I will be able to properly fix the issue? :)

tspopp, 2025-08-03:
One last bump before I close this due to inactivity.

closed 2025-08-03 by tspopp