tspopp/AquaMQTT #82: Thermor Airlis 270

batteoui, 2025-05-10:
Hello and thanks a lot for your valuable efforts
I've succesfully made the PCB and connected it to my Thermor Airlis 270 which has a fairly simple HMI. Couldn't get any data in the entities under HA. Seems like an unknown protocol so I flashed AquaMQTTDebug and could retrieve HEX and DEC files as attached.

Would be very grateful if you can have a look at those to have support for this Thermor Airliss 270 model.

Thanks again

[prefix_aquadbg_debug_dec.csv](https://github.com/user-attachments/files/20143654/prefix_aquadbg_debug_dec.csv)
[prefix_aquadbg_debug_hex.csv](https://github.com/user-attachments/files/20143655/prefix_aquadbg_debug_hex.csv)

tspopp, 2025-05-11:
Hey, unfortunately this serial protocol is not familiar to the ones we already have. I looked into the traces, and I see a lot of replayed data. I invested a little time and was not able to determine a message pattern, like id, length, checksums or even payloads. Also identifiers from the previous protocol are not in the traces, so this protocol must be something entirely new. 

```
2 186 214 88 100 255 213 56 128  
2 186 214 5 100 255 213 56 128  
2 186 214 5 100 255 213 56 128  
2 186 214 13 100 255 213 56 128  
2 186 214 13 100 255 213 56 128  
2 186 214 5 100 255 213 56 128  
2 186 214 88 100 255 213 56 128  
2 186 214 5 100 255 213 56 128  
2 186 214 5 100 255 213 56 128  
2 186 214 13 100 255 213 56 128  
...
```

So at this point I can't help or add anything to AquaMQTT. 

Does this heatpump has the same hardware as shown in [WIRING.md](https://github.com/tspopp/AquaMQTT/blob/main/WIRING.md)? Is the HMI looking the same? Or is it something else?

If you want to dig deeper and want to invest some time, you might want to get a logic analyzer and try to separate messages by date and time of dispatch. Maybe you can get an idea how these messages are built and what they contain.

batteoui, 2025-05-16:
Hello and thanks for your time.
Yes the heatpump contains similar hardware as in pictures attached. HMI is different tghough. I was able to captures some packets with a logic analyzer. I'm not a programmer so don't have a clue about what kind of protocol or baud rates is being used. I attach here in the captured frames with Saelae and Pulseview, maybe you can have a look at frames if you have some time ?

Thanks again.

[Capture.zip](https://github.com/user-attachments/files/20242107/Capture.zip)

[image]

[image]

[image]

batteoui, 2026-01-16:
Hello,
I realized I was not connecting the Ground pin on the logic analyze. I now have captured the correct signal that you can find attached. I exported the RAW data on csv. Apprently it decodes well with Asyn 9600 8N1 as I could retrieve he data. Could you please have a look at that capture if you find any relevance to the communication protocol ? 
Thanks a lot

[digital.csv](https://github.com/user-attachments/files/24673295/digital.csv)

tspopp, 2026-01-18:
Hey @batteoui, there is no data in these files. Maybe you attached the wrong file?