tspopp/AquaMQTT #83: Need for open collector transistors in the V.1 of circuit.

muh-1-1, 2025-05-11:
Hello,
Maybe "issue" is not appropriate section, but I do not see where I can simply put a question.
Why did you need to use transistors in the V.1 of the circuit? Aren't level converters "open drain" by themselves? At most, one could use pair of face-to face converters as described in Philips Application Note (http://cdn.sparkfun.com/tutorialimages/BD-LogicLevelConverter/an97055.pdf), page 12.
Cordially, Iouri

tspopp, 2025-05-12:
Maybe #36 and #33 will answer your question? :)

muh-1-1, 2025-05-12:
Sure, I understand the need for open collector on 1-wire line (as mentioned in #33). But my point is that, as far as I can see, level shifters already provide this functionality. But maybe not, I will think more about it. Did you try without the transistors? Or maybe you know a reason why level shifters are not enough? Also I am thinking about using even cheaper alternative to ESP32 S3. I have them, but I also have ESP32 C3, which should be adequate if memory suffices. Not pin-compatible, though. 
On the other subject, the mysterious 4-th pin (marked as unused in your docs) on my Thermor Aeromax 4 seem to be at steady 3 volts. Power for MCU?

tspopp, 2025-05-12:
Hm, honestly I never questioned this and treated those components isolated by their functionality. So I needed a LLC and I needed that circuit from the document for proper one-wire serial. But maybe you are on something and it can be dropped as well. I didn't read into the LLC specs.

In the beginning I used listener mode without the one-wire circuit. As soon as I tried MITM, I always had the one-wire circuit, just in case I send and receive on the bus at the same time by accident / during implementation. 

In regards to the C3: Currently I spawn the tasks for serial communication on CPU1, while keeping MQTT and communication on CPU0. As far as I know the C3 has a single core, so you might adapt the `spawn()` function of the tasks to change this. But I am not sure if it is still able to replicate the serial protocol while doing the other things. But I guess it is worth a test.

Well there are heatpumps which have only three wires, and some have four. My fourth wire does not show anything, so I guess it is left unused. 3V3 would be of course nice. But since there are a lot of different heatpumps, I wouldn't make a design decision to use that potential 3V3 line for power on a general purpose pcb ;).

Have you already tested AquaMQTT with the Aeromax 4, yet?

muh-1-1, 2025-05-12:
No, I have not tried yet. Just build ESP32 S3 version. C3 does not build, since Serial2 is not defined for C3. More strange is that USBSerial is not defined either. spawn() did not give me any build errors. 
Hope to test it tomorrow. It is sort of urgent, since in my aeromax display does not work and I hope to substitute it with AquaMQTT
I have noticed that there is difference in pins between ESP32 S3 and arduino nano. In configuration.h the pins are defined for ENV_DEVKIT_ESP32 as 5,6,7,8 and 2,3,4,5 otherwise. I guess, I will have 5,6,7,8, but I thought these two boards are pin compatible. Also 5,6,7,8 is not the default configuration for S3, but that is, probably, irrelevant.

muh-1-1, 2025-05-17:
OK. After several problems, it works in the listener mode now. 
The problems were: 1. wifi failure. Sometimes, sporadically, wifi becomes very bad. I have change one esp32 s3 for another and it is somewhat better, but still. Maybe just bad batch of modules. 
2. More interesting. I did without the transistors, just level shifters. But. (As Chat GPT told me) esp32 S3 in serial mode drives Tx _actively_. So it is not at all an open collector and can not be (AFAIK) made so in the serial mode. Was solved by adding a diode from the level shifter to ESP32. Also 2 chan level shifters I've got does not correspond to their schematic. Oh, well. 
I still can not make it work in MINM mode. It seem to me that AquaMQTT outputs to the same module it inputs from, instead of forwarding the packets, but I will look more carefully tomorrow. Just to make sure: is Serial N should be connected, both Rs and Tx to side N (as I understood from the docs and the code),  or Rx should be connected to N and Tx - to M? 
To summarize, Thermor Aeromax V4 seem to work, using Odysseus protocol.
Thank you for the software and idea. Great job. Although I found the code to be rather (or extremely) complicated for my amaterish mind.

tspopp, 2025-05-18:
> I have change one esp32 s3 for another and it is somewhat better, but still. Maybe just bad batch of modules.

That's why I went and recommend going for a high quality one, the Arduino Nano ESP32 is really top notch. No issues here, did not waste any time in debugging strange wifi or other stability issues.

> More interesting. I did without the transistors, just level shifters.

I put them there to be safe, not accidental sending and receiving on the same wire at the same time. But in LISTENER mode everything should be safe anyway.

> Just to make sure: is Serial N should be connected, both Rs and Tx to side N (as I understood from the docs and the code), or Rx should be connected to N and Tx - to M?

Yes, Serial1 RX and TX to the one-wire HMI and Serial2 RX and TX to the one-wire Serial2

> To summarize, Thermor Aeromax V4 seem to work, using Odysseus protocol.

Nice, I will put that on the list. The ODYSSEE protocol has been added very recently. Just let me know if you come across any bugs related to the protocol.

> Although I found the code to be rather (or extremely) complicated for my amaterish mind.

Well, code written by someone else sometimes feels complicated. I admit this thing grown over time, but I really don't see the need to put more time into this since this project is rather complete - except that I add more heatpumps if their protocol is familiar to the ones I have. Apart from that, the code is maintainable and supports multiple protocols and there are currently zero bugs known. So I do not share the impression of overly complicated code :) But If you feel that way, I am always happy to look at PRs contributing to this project.

muh-1-1, 2025-05-18:
> I put them there to be safe, not accidental sending and receiving on the same wire at the same time. But in LISTENER mode everything should be safe anyway.
I think, my circuit has no problem surviving two sides pulling the line down at the same time. 
> es, Serial1 RX and TX to the one-wire HMI and Serial2 RX and TX to the one-wire Serial2
OK. I still do not understand why ControllerTask seem to be talking and listening to Serial2 and HMI tasks deals only with Serial1. 

Meanwhile, I have resolved my main problem - persistent beeps after heating is finished. From error debug messages it is error 15 - error of date/time. No wonder, since the display does not work and I can not set the datetime.  But no real problem. One thing - I see the error code in error debug raw, but not in aquamqtt/main/errorCode. Ah, I see. It is not supported with Odyssee. It is there, though.

muh-1-1, 2025-05-18:
Two more things: 1. I can not get output from the serial. Tried all combinations of CDC_ON_BOOT etc, nothing. log_e works, though, but it code is not available, I think. 
2. In any case, since the device is embedded, it is more practical to have telnet debug output, which I have added. It works, but it seem that if there is too much output (like raw serial data), MCU reboots.

tspopp, 2025-05-18:
On legacy heatpumps error 15 is a pure HMI error which is not transferred in the serial protocol. Not sure if this is the same for Odyssee heatpumps.

tspopp, 2025-05-18:
Serial0 is used for any debug outputs. Make sure not to output anything on Serial1 and Serial2 used by one wire communication

muh-1-1, 2025-05-18:
Error 15 is described on Thermor site. Of course, they mean number displayed on their display, but it is, probably, the same as in communication. I do not know whether the error message comes from the main or hmi, it is what I see in MQTT as errordebug.
I do have problem with MITM. As soon as I connect the units, HMI begins to spit out long messages, but the main only sends 1 byte in 67 217 193 194  sequence 100 ms apart. As I understand, since it is only 1 byte, the software never recognize it as a complete frame and so `mSequence` is never updated. 
        if (const int result = mBuffer.pushByte(valRead); result > 0) {//muh
            mSequence.pushOverwrite(result);
        }
Thus, the main never sends any packets, since it expects `mSequence` to have specific values. I do not know whether other mains behave differently or there is some problem with my hardware setup (seem to be working OK, I checked with an oscilloscope) or I misunderstood the code. I've tried to send194 no matter what mSequence is - still no go. But why the main send these 67  217 193 bytes? Could it be that in Odyssee hmi is supposed to finish them all?

tspopp, 2025-05-18:
Can you share some raw serial traces? The only heat pump which uses odysee protocol send messages in the pattern 67 - 217 - 193 - 194 - 74. Also see the protocol documentation here: https://github.com/tspopp/AquaMQTT/blob/main/PROTOCOL_ODYSSEE.md

If your heatpump is using a different pattern, we need to adapt it accordingly. 

In the communication within the main controller, we expect a magic byte 194 when its time to send and complete the HMI message on the bus. To ignore any other possible bytes which might be valid 194, it uses a sequence detection. So only in case we received the main messages in the expected pattern, we treat the byte "194" as magic byte and take action.

muh-1-1, 2025-05-18:
The main sends bytes in the same sequence. But just bytes, not messages. No only 194 but 67-217-193-194. 
Serial data - my pleasure. Should I just them here? How many?

tspopp, 2025-05-18:
I guess 10 seconds of data is enough:) Just add it here.

muh-1-1, 2025-05-18:
I think now, that the problem with MINM is in my setup. The main seem to be very, very finicky about logical levels. I have connected two diodes instead of passthrough, so 0 is shifted to 0.6 volts, and the communication is broken. The main sends the same sequence, the HMI sees 194 and responds (apparently +0.6 instead of 0 is not a problem for the HMI). But the main keeps sending those single bytes. 
Anyway, here are the serial data. Listener mode, regular passthrough.
sensor.hmidebug,1D00100010000000000000000000004000211E390E0000000054430000,2025-05-18T15:03:05.674Z
sensor.hmidebug,1D00100010000000000000000000004000211E3A0E0000000054430000,2025-05-18T15:04:05.733Z
sensor.hmidebug,1D00100010000000000000000000004000211E3B0E0000000054430000,2025-05-18T15:05:05.763Z
sensor.hmidebug,1D00100010000000000000000000004000211E000F0000000054430000,2025-05-18T15:06:05.807Z
sensor.hmidebug,1D00100010000000000000000000004000211E010F0000000054430000,2025-05-18T15:07:05.854Z
sensor.hmidebug,1D00100010000000000000000000004000211E020F0000000054430000,2025-05-18T15:08:05.301Z
sensor.hmidebug,1D00100010000000000000000000004000211E030F0000000054430000,2025-05-18T15:09:05.354Z
sensor.hmidebug,1D00100010000000000000000000004000211E040F0000000054430000,2025-05-18T15:10:05.392Z
sensor.hmidebug,1D00100010000000000000000000004000211E050F0000000054430000,2025-05-18T15:11:05.442Z
sensor.hmidebug,1D00100010000000000000000000004000211E060F0000000054430000,2025-05-18T15:12:05.486Z
sensor.hmidebug,1D00100010000000000000000000004000211E070F0000000054430000,2025-05-18T15:13:05.535Z
sensor.hmidebug,1D00100010000000000000000000004000211E080F0000000054430000,2025-05-18T15:14:05.579Z

sensor.maindebug,1D3A00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:03:58.414Z
sensor.maindebug,1D3B00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:04:18.829Z
sensor.maindebug,1D3A00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:04:19.436Z
sensor.maindebug,1D3B00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:04:23.024Z
sensor.maindebug,1D3A00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:04:23.626Z
sensor.maindebug,1D3A00160015001500160000003A002C00400000000002000000005243,2025-05-18T15:04:36.835Z
sensor.maindebug,1D3A00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:04:38.045Z
sensor.maindebug,1D3A00160015001500160000003A002C00400000000002000000005243,2025-05-18T15:04:44.648Z
sensor.maindebug,1D3A00160015001400160000003A002C00400000000002000000005243,2025-05-18T15:04:50.646Z
Only one energydebug and no errordebug
sensor.energydebug,19490262200000000049020000622000000000000000000000,2025-05-18T14:57:33.637Z

muh-1-1, 2025-05-20:
OK. I redid my circuit, putting 7207 OC buffer instead of a level shifter. It still does not work as MITM. Also the main has changed, at some point (not necessarily when I've changed the circuit), its behavior. Now it send full frames for 67...193 not waiting to establish connection with the Hmi. Before it was sending only single bytes and started sending full frame only when HMI talked back. In MITM mode they both transmit frames, not synchronizing between themselves. Could it be delay between main's 194 byte and the response from AquaMQTT? In passthrough mode it is quite short, ~100 usec, but with AquaMQTT it can be almost 10 msec. I will play a little more, but with less urgency. 
BTW, I did not ask - the overrides are temporary? If AquaMQTT is not there or is in the listener mode, they revert to whatever is memorized in the HMI? Or is this information memorized in the main? I need to change air duct configuration, do not know how important it is.

muh-1-1, 2025-05-20:
Ah. Maybe it is OK. They are not supposed to be synchronized, are they. So maybe it works (with new main's behaviour). But number of MainBytesDropped is increasing.

tspopp, 2025-06-10:
@muh-1-1 Can I close this? As far as I understood the issues were related to your custom hardware design?

muh-1-1, 2025-06-10:
Sure. Any progress on ODYSEE protocol? In particular - any way to reset the alarm, like in the legacy protocol? Also, it would be useful, maybe, to request errors from the main. I implemented that, but only for error No 1 and in a klutchy way. Another thing - would it not be nice to set defaults for override item. Again, I've implemented that, since HMI thinks that my duct config is Int/Ext and it is Int/int. HMI display is not working, so I can not reconfigure it in a regular way. If one wants to override something, he probably wants to override it always, not restore the override each time aquamqtt is reset. This can be done in the code (like I did) or by writing the overrides in "preferences" arduino storage.

tspopp, 2025-08-03:
Hey, since I don't have an ODYSSEE heatpump - there will be no more progress from my side as long as nobody is providing input/feedback. All other items are related to a feature request: using AquaMQTT without an HMI controller. This is something which is feasible but currently not on my short term to do list. I created a ticket for this in #87.

closed 2025-08-03 by tspopp