tspopp/AquaMQTT #86: Comfort by Sanibel CV5

nitriboz, 2025-08-01:
Hello tspopp,
first of all, i would like to thank you for your work in this project and your ongoing support.
I got a HeatPump installed two weeks ago. In the last days i tried to install the aquamqtt board (rev 2) in combination with ESP32-S3-Nano Development Board to include the infos in my iobroker installation.
Flashing the ESP worked after several attempts because of some driver issues (using Ardunio IDE to recognize the ESP fixed it).
 
The **model** is called **"Comfort by Sanibel CV5"**
("Comfort by Sanibel" is some kind of reseller brand of a german wholesale company for plumbing and heating. My Plumber said that this model is based on the "Austria Email Explorer EVO2" - looking a bit different but the manual is pretty similar.)
instructions/manual, if it's of any interest: [PDF]([https://comfort-by-sanibel.de/profibereich/?oxDeeplink=eyJuYXZQcm98MSI6eyJ2aWV3IjoiZmlsdGVyZWQiLCJmaWx0ZXJzIjp7InByb2R1Y3QtdHlwZSI6WyJQUk9EVUNUIl0sInNlcmllcyI6WyIzTzA2VU1RQTBITTFOUVU5QkNVSTdHRzBMTyJdfSwicXVlcnkiOiIiLCJoaWRkZW5GaWx0ZXJzIjp7ImJyYW5kIjpbIkpDMkJTSUROMThMMjhFUk5FM1FGTjVTSjkwIl19fSwib3ZlcmxheSI6eyJ0eXBlIjoib3BlbkRvY3VtZW50IiwiYXJncyI6eyJkb2N1bWVudCI6IjEwNTk5MDQ2IiwicGFnZU51bWJlciI6MSwic2luZ2xlUGFnZSI6dHJ1ZSwicXVlcnkiOiI2MTA3NTUwIE9SIDM0MTA1Mzk4NjE1MTAiLCJleHRlcm5hbExpbmtzRW5hYmxlZCI6dHJ1ZSwicGFnZSI6MX19LCJuYXZQcm98MiI6eyJ2aWV3IjoiZmlsdGVyZWQiLCJmaWx0ZXJzIjp7fSwicXVlcnkiOiIiLCJoaWRkZW5GaWx0ZXJzIjp7ImJyYW5kIjpbIkpDMkJTSUROMThMMjhFUk5FM1FGTjVTSjkwIl19LCJoaW50IjoiUHJvZHVjdHMifX0%3D](https://deref-gmx.net/mail/client/KXJ6_S3dsDg/dereferrer/?redirectUrl=https%3A%2F%2Fcomfort-by-sanibel.de%2Fprofibereich%2F%3FoxDeeplink%3DeyJuYXZQcm98MSI6eyJ2aWV3IjoiZmlsdGVyZWQiLCJmaWx0ZXJzIjp7InByb2R1Y3QtdHlwZSI6WyJQUk9EVUNUIl0sInNlcmllcyI6WyIzTzA2VU1RQTBITTFOUVU5QkNVSTdHRzBMTyJdfSwicXVlcnkiOiIiLCJoaWRkZW5GaWx0ZXJzIjp7ImJyYW5kIjpbIkpDMkJTSUROMThMMjhFUk5FM1FGTjVTSjkwIl19fSwib3ZlcmxheSI6eyJ0eXBlIjoib3BlbkRvY3VtZW50IiwiYXJncyI6eyJkb2N1bWVudCI6IjEwNTk5MDQ2IiwicGFnZU51bWJlciI6MSwic2luZ2xlUGFnZSI6dHJ1ZSwicXVlcnkiOiI2MTA3NTUwIE9SIDM0MTA1Mzk4NjE1MTAiLCJleHRlcm5hbExpbmtzRW5hYmxlZCI6dHJ1ZSwicGFnZSI6MX19LCJuYXZQcm98MiI6eyJ2aWV3IjoiZmlsdGVyZWQiLCJmaWx0ZXJzIjp7fSwicXVlcnkiOiIiLCJoaWRkZW5GaWx0ZXJzIjp7ImJyYW5kIjpbIkpDMkJTSUROMThMMjhFUk5FM1FGTjVTSjkwIl19LCJoaW50IjoiUHJvZHVjdHMifX0%253D))  (at the declaration of confirmity, there is an advice to the ATLANTIC Group)
The connection cable has only three wires (red, white, black) and the plug at the HMI-side is only "one row", so i helped me to use some "GPIO cables" to avoid cutting of the original connection cable.
 
1. Try: flash with MITM mode
   --> MQTT Broker got basic infos (like "aquamqtt.stats.aquamqttMode=MITM"), but nothing out of the heatpump itself.
   Connecting the aquaMqtt board between the DHW main board and the DHW HMI board leads to the emergency mode of the heat pump (operation with heating element only).
2. Try: flash with LISTENER mode 
   --> same result, mqtt.0.aquamqtt.stats.aquamqttMode = LISTENER instead of MITM
3. Try: flash aquaDebug
   --> same result at the heatpump, got almost empty result files from aquaDebug, but i am not sure, if i flashed aquaDebug right, because i hab a bit difficulties in understanding the python commands. Is anywhere an more beginner friendly description available?
4. Try: only "bridging" the original plug at the HMI-side with three single wires without using the aquaMqtt-Board to exclude the wiring material as source of error.
   --> Heat pump works in normal mode
 
Do you have any idea what i can try next?
My approach would be to understand the AquaDebug installation instructions and hope to get useful data for further analysis.
Or can i try an other protocol? I did not see yet, where i could set the protocol type. Or is the protocol type recognized automatically?

Emergency Mode: 
[image]

Click "OK" once: 
[image]

Click "OK" again: 
[image]

HMI - backside infos:
[image]

Mainboard:
[image]

mainboard connector ("A3", black cable with black, white, red wires)
[image]

wiring
[image]

wiring detail
[image]

tspopp, 2025-08-01:
Hey, sorry I got little time today, but If you want to use AquaDebug or AquaMQTT in Listener mode (which is the first logical thing to test) then you must shorten or set the Passtrough (PSTRH) on the board e.g. using a jumper or a wire. In this setup, heatpump is expected to work as expected, but you should have the possibility to analyze/capture the data. If this is still NOT the case and you are not able to capture data with AquaDebug, this heatpump is using different baud rates or a different serial protocol. In this case I would go for an cheap logiz analyzer e.g. from AZDelivery to check what is actually on the wire.

I try to answer any other question as soon as I have time. You may also join the discord chat for debugging ;)

nitriboz, 2025-08-02:
I don't get the Listener Mode really to work. I activated some debug options in the aquaMqtt config files (DEBUG_RAW_SERIAL_MESSAGES, DEBUG_PUBLISH_DROPPED_MESSAGES, MQTT_PUBLISH_SERIAL_STATISTICS).
Now, there are a lot of dropped Bytes since yesterday evening (round about 22:50).

And i set MQTT_FULL_UPDATE_MS to one minute, but that was obviously without any effect, because the messages can not get processed.

So i sent you the csv files via discord. Thanks for your support.

[image]

tspopp, 2025-09-20:
Few weeks ago, I made some progress with @nitribo, discussion happened via discord. In my branch cv5-modbus, there is a prototype project called [CV5Listener](https://github.com/tspopp/AquaMQTT/tree/t/cv5-modbus/prototypes/CV5Listener) which is capable of dumping emitted messages and providing them raw - without any interpretation - to MQTT. You have to set the AquaMQTT board to LISTENER mode (jumper), adapt the configuration (wifi, mqtt broker, etc.) of the prototype and flash it.

[image]

A message has several header parameters which is reflected by the MQTT topic, while the raw data is posted on the topic `debug` as hex-string: `functionCode/parameter01/parameter02/subfunction/debug`. You can use the [python script](https://github.com/tspopp/AquaMQTT/blob/t/cv5-modbus/prototypes/CV5Listener/collect.py) to create message dumps of the communication, which might help you in identifying values.

Apart from that there are a lot of unknowns and currently I hardly believe that I will be able to do the implementation, due to lack of hardware, time and interest. Moreover, somebody has to understand very closely how the protocol works, for example which messages are emitted by the main controller, which by the hmi controller. And only as soon as there is a clear understanding how the protocol works , you can even think about replicating everything via MITM. But until then, the best thing you can do is trying to identify values in the data dump, and implementing a very lightweight listener mode. 

So if you want to help, you can try identifying values in the raw data:

As an example, this has been provided by @nitriboz:

Topic: `100/254/176/7`

Message dump in decimal representation, generated by the python script:
```
2025-08-11 19:29:56.742203,1 100 254 176 7 14 19 67 28 189 7 229 4 230
2025-08-11 19:29:57.421918,1 100 254 176 7 14 19 67 28 186 7 229 4 230
2025-08-11 19:29:58.108124,1 100 254 176 7 14 19 70 28 178 7 227 4 230
```

Interpretation:
- Byte 0 - 5: Fixed Header (Slave `1`, FunctionCode `100`, Parameter01: `254`, Parameter02: `175`, SubFunction `7`, TotalLength `14`) 
- Byte 6 - 7: `19 70`. Interpreted as uint16 (Calc: `256*19+70 == 4934`) --> Water Temperature (Calc: 4934 / 100 = 49,34°C)
- Byte 8 - 9: `28 178`. Interpreted as uint16: `7346`,  --> Compressor Outlet Temperature (Calc: 7346 / 100 = 73,46°C)
- Byte 10 -11: `7 227` Interpreted as uint16: `2019` --> Supply Air Temperature (Calc: 2019 / 100 = 20,19°C)
- Byte 12 - 13: `4 230`: Yet Unknown.