tspopp/AquaMQTT #92: Bug: doesn't heat anymore after some time

NickdeK, 2025-09-09:
So this is a weird issue that is hard to reproduce and debug. I've been holding off on creating an issue to collect some more information about this, unfortunately without much result.

[image]

It seems that after 1 or 1,5 months the device stops warming up. The device is set in a timed window mode. Changing settings in the AquaMQTT integration doesn't seem to fix the timer issue. The only way to fix this is to power flip the AquaMQTT. For me this is happening for some time basicly in the same interval, but this screenshot shows the issue the best (vacation muddles a total overview).

On a Dutch forum (Tweakers) there have been multiple messages about this issue:
https://gathering.tweakers.net/forum/view_message/83367010
https://gathering.tweakers.net/forum/view_message/82908208
https://gathering.tweakers.net/forum/view_message/82908890
https://gathering.tweakers.net/forum/view_message/82997622
https://gathering.tweakers.net/forum/view_message/83038872

And there is a google sheet with a overview of users experiencing all kind of problems, but some clear "time" issues: https://docs.google.com/spreadsheets/d/1fYkOCbF5XYuxVKj7GcjqPrY7jN1sAYLELUVb9Mwt4uY/edit?gid=442106297#gid=442106297

What information can I (or we, if I can get the Tweakers topic to help here) supply to debug this issue?

tspopp, 2025-09-09:
Interesting, if something is wrong in timer mode it is likely due to an issue with time and date. I have a few questions to narrow it down:

- Are you using the latest version of AquaMQTT? Asking, since there were a few fixes in regards to time and date in the more recent versions.
- Instead of power flipping AquaMQTT, have you tried forcing "ALWAYS ON" mode? Will it heat again? What happens if BOOST is toggled through AquaMQTT, will it heat up properly in this faulty state?

To troubleshoot:

- Disable time and date overrides in configuration. I bet without this override this issue will be away. In this case the official HMI Controller is the time master again and a fresh booted AquaMQTT just forwards messages, even in MITM mode.
https://github.com/tspopp/AquaMQTT/blob/b9317c1a8ef49687d6b6e119d3548a1652b1ba36/AquaMQTT/include/config/Configuration.h#L52-L55
- Enable time and date debugging and see if these values change to something weird when the issue occurs:
https://github.com/tspopp/AquaMQTT/blob/b9317c1a8ef49687d6b6e119d3548a1652b1ba36/AquaMQTT/include/config/Configuration.h#L97-L103

NickdeK, 2025-09-10:
> Are you using the latest version of AquaMQTT? Asking, since there were a few fixes in regards to time and date in the more recent versions.

I'm running the latest version of AquaMQTT.

> Instead of power flipping AquaMQTT, have you tried forcing "ALWAYS ON" mode? Will it heat again? What happens if BOOST is toggled through AquaMQTT, will it heat up properly in this faulty state?

With regards of trying ALWAYS ON; I have seen one message on the previously mentioned forum that says that it starts warming up again when going to ALWAYS ON here: https://gathering.tweakers.net/forum/list_message/83367010#83367010 When I tried that I think it didn't work, but I might be mistaken on that.

Next time it happens I will try both BOOST mode first and ALWAYS ON afterwards to see what happens.

> Enable time and date debugging and see if these values change to something weird when the issue occurs:

I will enable this, but this will reset the AquaMQTT and I think it will take at least a month before I can share the results.

Thinking about this a bit more, I was wondering if this maybe is a timezone issue? I noticed before installing the AquaMQTT that the time on the V4 was off by one hour in winter time or summer time (not sure which one). The winter time will be set here on the 26th of October (Central European Time, GMT +1), summer time will be set on 29th of March (Central European Summer Time, GMT +2).

I don't have the AquaMQTT device installed long enough to be sure if this might be the cause. It's just something I noticed earlier.

tspopp, 2025-09-10:
> Thinking about this a bit more, I was wondering if this maybe is a timezone issue? I noticed before installing the AquaMQTT that the time on the V4 was off by one hour in winter time or summer time [...]

By default it is using the timezone pre-configured and it should handle summer / winter time as well. You may want to edit the timezone and validate if the time and date is correct (as soon as you have enabled publishing time and date to MQTT).

https://github.com/tspopp/AquaMQTT/blob/b9317c1a8ef49687d6b6e119d3548a1652b1ba36/AquaMQTT/include/config/Configuration.h#L25-L29

MuadDib007, 2025-10-29:
Well, time to enable debug as since the heating cycle from last night, the system decided to heat up again this morning outside of the time window.

So now I have power cycled the heatpump and it is working as expected again.
Time to enable the debug logging.

Time setting on the heatpump itself is OK btw...

NickdeK, 2025-11-05:
> constexpr bool DEBUG_RAW_SERIAL_MESSAGES = true;

Unfortunetly this is not enabled. Anyway to enable this without actually flashing and rebooting the device?

> constexpr bool OVERRIDE_TIME_AND_DATE_IN_MITM = true;

This is indeed set to true!

NickdeK, 2025-11-05:
Sorry for the many messages!

I was thinking I might use claude to change the code and change the lastStateUpdate time so that the bug occors in a few minutes to validate if this is actually the bug. Afterwards I can apply the suggested fix and also validate if that works and create a PR.

Let me know if you think that would be fine. I don't want to reset the bugged state the device is in right now and miss possible opportunities to debug this.

tspopp, 2025-11-07:
Yes, sure! 

Maybe the AI is right, but I have to look into it. But my assumption goes into the same direction, I guess there is something broken in the AquaMQTTs time master implementation. So in the meantime you should get rid of the error, by disabling the time overrides in the config.

Of course I am happy if you make an proposal for the fix. In any way, my next time slot is reserved for looking into this :) Thanks for debugging and sharing the Information.

NickdeK, 2025-11-07:
Well good news! I could reproduce the issue easily and fixed it with a single line change. The PR is ready for review #99 :)