tspopp/AquaMQTT #96: Pre-assemble issue through JLCPCB

merijnbos81, 2025-09-25:
Today I tried to order a pre-assembled pcb for revision 2.0. This is the first time I have ever done this, so I carefully followed the instructions. All seemed fine but now I received an e-mail from JLCPCB that something is wrong with the slot size being to small.

"Sorry to bother you, but we'd like to confirm with you an issue found when engineer making production file of your PCB order.

As shown below, the indicated slots are only 0.5mm which is less than the minimum slot 0.8mm we can support, shall we cancel the order for refund for you to change the file and reorder with correct file? Since the PCBA order doesn't support file-replace."

[image]

Did I do something wrong, or is there really an issue with the design of the slots?

merijnbos81, 2025-09-26:
I mentioned that the identical design has succesfully went through earlier. Now they are asking for the ordernumber so the engineer can use it for reference. Any chance you can find it and share it? Many thanks!

tspopp, 2025-09-26:
W202503130230578

merijnbos81, 2025-09-26:
Thanks! They responded that they will pass it on to the engineer. I will let you know if it's successful.